ARMmbed/DAPLink #282: Daplink did not flash correct data if flash algo > 0xA00 size

maclobdell, 2017-07-12:
While testing Daplink with an updated flash algo for NCS36510, it was flashing all 00's to the target flash instead of the application.

I noticed that there was a large section of 00 data in the flash blob.  This was due to a large array in the flash algorithm that was declared but never used.  This large array caused the flash algo to go over the 0xA00 offset that is hard coded to be the start of the RAM memory buffer.  

```
    0x3fff4000 + 0x00000A00,  // mem buffer location

```
 
I suspect that daplink was finding the data which was all 00s in the flash blob and programming that instead of the real data.  

Perhaps the flash blob is getting updated in the ram at each page program and thus the real buffer data is getting over written?  

See attached working and non working flash blob files which demonstrate the problem.  

[flash_blob_buffer_broken.c.txt](https://github.com/mbedmicro/DAPLink/files/1143171/flash_blob_buffer_broken.c.txt)
[flash_blob_nobuffer_functional.c.txt](https://github.com/mbedmicro/DAPLink/files/1143173/flash_blob_nobuffer_functional.c.txt)

c1728p9, 2017-07-12:
Yep, turns out DAPLink is actually re-writing the entire flash algorithm into ram on every call. This issue is fixed by #283 which removes the extra re-writing of the flash algorithm.

c1728p9, 2017-11-21:
Closing this issue since the fix has been merged.

closed 2017-11-21 by c1728p9