ARMmbed/DAPLink #284: USB endpoint calculations check issue

awais-ublox, 2017-07-17:
The distinct endpoint validation check for HID and MSC interface requires fixing, as USBD_HID_EP_INTIN is not compared with **USBD_MSC_EP_BULKOUT**

For file(s) usb_config.c in hic_hal:

#if ((((USBD_HID_EP_INTIN   == USBD_MSC_EP_BULKIN)  || \
       (USBD_HID_EP_INTIN   == **USBD_MSC_EP_BULKIN**)))|| \
      ((USBD_HID_EP_INTOUT  != 0)                   && \
       (USBD_HID_EP_INTOUT  == USBD_MSC_EP_BULKIN)  || \
       (USBD_HID_EP_INTOUT  == USBD_MSC_EP_BULKOUT)))
#error "HID and Mass Storage Device Interface can not use same Endpoints!"

c1728p9, 2017-11-21:
Thanks for reporting this @awais-ublox. I created a PR https://github.com/ARMmbed/DAPLink/pull/352 to fix this.

awais-ublox, 2017-11-22:
You are welcome :)

c1728p9, 2017-11-28:
Closing since the fix was merged

closed 2017-11-28 by c1728p9