ARMmbed/DAPLink #285: Format the disk in drive E: before you can use it

vizzupotluri, 2017-07-19:
I am trying to flash the nrf52 DK using lpc11u35. When I am trying to plug the usb into PC it is showing the above mentioned error message i.e. "Format the disk in drive E: before you can use it"

Can anyone please let me know,What may be the problem is?

c1728p9, 2017-07-22:
Hi @vizzupotluri, what firmware is running on the lpc11u35? Is it from a recent version of DAPLink? Could you attach a copy of it?

vizzupotluri, 2017-08-09:
I am using IDAPLink firmware file which was modified by me for nRF 52 which was orginally created by  Nguyen Hoan Hoang. Please feel free to go through the link of Github repo which was attached below[https://github.com/I-SYST/CMSIS-DAP_IDAP-Link](url)

cederom, 2017-08-09:
Hello @vizzupotluri, please note we can help you only with the offical DAPLink release and codebase.. Can you check/provide following information:
* what is the exact name and model of the target board?
* what is the exact name of the target MCU?
* do you use onboard debugger or your own external board?
* if you use external debug board could you please give more details on hardware?
* have you tried updating firmware on your DAPLink debugger board to the last official release (v0234)?
* have you tried connecting DAPLink to a different computer and operating system? does problem persist?
* have you tried another USB cable (preferably from the certified source/vendor)? does problem persist?

c1728p9, 2017-09-25:
Closing due to inactivity. Please re-open if you have further questions or need assistance.

closed 2017-09-25 by c1728p9