ARMmbed/DAPLink #291: Does not support stm32c8t6 firmware？

panrenjie, 2017-07-23:
in the hardware project mbed-HDK， I find  INTERFACE DAPLink with stm32c8t6 ,but I can't find the  firmware for it ?  So ,I want to know is there any firmware that supports stm32c8t6?

909958846, 2017-07-27:
I also want to know how to build hex file for stm32f103c8t6

cederom, 2017-08-22:
Hello @panrenjie, please precise what do you mean by `stm32c8t6`, is it `stm32f103c8t6` as stated by @909958846 ? Do you want to use this chip as Target you want to debug or HIC you want to run DAPLink on it?

cederom, 2017-08-22:
@panrenjie @909958846 , there may be an approach for HIC on this chip! have you tried https://github.com/mbedmicro/DAPLink/pull/133? You would be best people to try it out and let us know how it works :-)

brianesquilona, 2018-12-28:
Stm32f103xb as hic is supported by #257

closed 2018-12-28 by brianesquilona