ARMmbed/DAPLink #309: Add FreeBSD support

cederom, 2017-08-28:
At the moment FreeBSD support is missing for DAPLink, below is the TODO list:
* [ ] add support for mbedls [mbedls].
* [ ] research and document working `automount` configuration for CI testing at user level (FreeBSD 10.2+) [freebsd_automount].
* [ ] research and leverage innovative USB stack [freebsd_usb], `usbconfig` [freebsd_usbconfig] and `usbdump` [freebsd_usbdump] utilities for USB traffic dump/analysis and DAPLink configuration/control.
* [X] UMS support.
* [X] CDC support.
* [X] HID support.

[mbedls] https://github.com/ARMmbed/mbed-ls
[freebsd_automount] https://www.freebsd.org/doc/handbook/usb-disks.html
[freebsd_usb] https://www.freebsd.org/doc/en/books/arch-handbook/usb.html
[freebsd_usbconfig] https://www.freebsd.org/cgi/man.cgi?query=usbconfig&sektion=8
[freebsd_usbdump] https://www.freebsd.org/cgi/man.cgi?query=usbdump&sektion=8

carlosperate, 2017-12-18:
Was this close because FreeBSD support was dropped, or because it should be working now?

hselasky, 2017-12-19:
Hi,
With the MicroBit I got my hands on, at least uploading the .hex files was not working, because FreeBSD was writing the file data out of order apparently. Mounting the partition with "-o sync" and then "dd'ing" the file to a location inside the partition resolved this. I also I noticed using a large filename with spaces and stuff also confused the MicroBit.
--HPS

cederom, 2017-12-23:
Hey there! This issue was closed because it was considered not welcome/priority by the project.. however we re-discussed that recently.. maybe it will be reconsidered at least min priority ;-)

Since I am not the only one anymore who wants it working I will reopen the issue and add support eventually in my free time because FreeBSD is the best and I love it too :-)

HPS I may have some questions for you regarding automounter and maybe other USB related configurations for FreeBSD.. any support from you would be highly appreciated so we could get things done quickly! :-)

DAPLink uses FAT12 so only the short names are supported. Out of order write yes is a problem for BIN files but should not be for HEX files except HEX is really fragmented.. we can verify that while porting.. In case of additional issues please report them separately on GitHub :-)

MERRY CHRISTMAS! :-)

c1728p9, 2017-12-24:
> DAPLink uses FAT12 so only the short names are supported.

Actually DAPLink uses [FAT16](https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/drag-n-drop/virtual_fs.c#L150). It still doesn't support long filenames though.

> Out of order write yes is a problem for BIN files but should not be for HEX files except HEX is really fragmented.

Hex file parsing only works when the file is sent sequentially. If a hex file is sent out of order DAPLink aborts programming, as the parser will not function correctly.

hselasky, 2017-12-24:
Hi, the smallest size that will be written is given by the block size. First the block size of the disk 512 bytes typically and then the block size of the file system. If the Mirobit would format the file system using a block size of 8MBytes for example, re-ordered writes would not be any problem. Another idea is to embed an offset counter every 512 or 4K or whatever, which the Microbit will use to identify out of order writes. What do you think?   Further it would be cool if the hex file could simply be dd'ed to the beginning of the disk, and the MicroBit would understand that.

c1728p9, 2017-12-24:
Hi @hselasky, I'm a bit surprised you are getting out of order file transfers in the first place. How were you copying the file when you saw this? I have seen a similar problem to this on Linux, but in that case it was due to accidentally having multiple mount points for the same device. Think anything like that may be happening?

>  If the Mirobit would format the file system using a block size of 8MBytes for example, re-ordered writes would not be any problem.

I hadn't thought about that. Unfortunately, after looking at the FAT specification I found the maximum sector size allowed for FAT is 4K. I think there may also be compatibility issues if this is set to a value other than 512.

> Another idea is to embed an offset counter every 512 or 4K or whatever, which the Microbit will use to identify out of order writes. 

This is definitely a possibility. The problem with this though is DAPLink will need a more complex mechanism to tell which sectors have been erased and programmed which will take more ram. When a file is sent sequentially this is much simpler and doesn't require much ram. Another problem that would need to be avoided is re-programming the same sector multiple times, as sometimes the host PC sends the same sector multiple times.

hselasky, 2017-12-25:
Hi, I think that this is normal. The SCSI layer does not always understand if the underlying media is a spinning disk or flash disk.  In case of spinning disk the SCSI layer tries to optimise the writes according to the physical location on the platter, so that the movement of the disk head becomes less. This can result in out of order writes.

I'm worried that meta-data writes, like updating the FAT tables, might cause out of order writes.

That FAT has some info at the beginning:

     struct bsbpb {
         uint16_t    bpbBytesPerSec;         /* [-S] bytes per sector */
         uint8_t     bpbSecPerClust;         /* [-c] sectors per cluster */
         uint16_t    bpbResSectors;          /* [-r] reserved sectors */
         uint8_t     bpbFATs;                /* [-n] number of FATs */
         uint16_t    bpbRootDirEnts;         /* [-e] root directory entries */
         uint16_t    bpbSectors;             /* [-s] total sectors */
         uint8_t     bpbMedia;               /* [-m] media descriptor */
         uint16_t    bpbFATsecs;             /* [-a] sectors per FAT */
         uint16_t    bpbSecPerTrack;         /* [-u] sectors per track */
         uint16_t    bpbHeads;               /* [-h] drive heads */
         uint32_t    bpbHiddenSecs;          /* [-o] hidden sectors */
         uint32_t    bpbHugeSectors;         /* [-s] big total sectors */
     };

Do you know what are these values for MicroBit. I don't have my MicroBit at hand right now.

I suspect that if the number of "drive heads" is one, the problem will go away.

cederom, 2017-12-25:
Hello @c1728p9 @hselasky thanks for feedback! :-)

True, FAT16 is used my mistake, still 8.3 filenames required.. maybe we can somehow painlessly switch to FAT32 or at least truncate long names @c1728p9 ?

@hselasky here are sources of VFAT implementation:
* USB MSC handling: https://github.com/ARMmbed/DAPLink/tree/master/source/usb/msc
* SCSI + Flash handling: https://github.com/ARMmbed/DAPLink/tree/master/source/daplink/drag-n-drop

Here are parameters that you mention, looks like drive heads is one already..
https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/drag-n-drop/virtual_fs.c

hselasky, 2017-12-25:
OK, cool. I need to think about my priorities, but for sure we could make a patch there. It is just plan C-code :-)

cederom, 2017-12-26:
thanks @hselasky :-) the drag-n-drop flashing out-of-order-write is not that easy to fix, it depends strongly, but not only, on underlying target flash organization/access/write/erase and DAPLink supports lots of different targets. linear access is the simplest and effective way to achieve the goal sometimes on MCU that has only 16KB of RAM.. unfortunately we need to comply to the weakest HIC..

my initial goal for FreeBSD support was to run CI Firmware and Builds Automation Tests on FreeBSD machine not the Linux one, but that would also bring the FreeBSD as the supported platform for ARM mbed and Microbit..

i would bring the basic  OS support in the first place, automounter configuration and the mbedls utility port, so we can run the automation tests for various boards. i can take care of mbedls. help with automounter would be appreciated as no partition table seemed to cause problems but that seems fairly easy and someone on the mailing list already replied the solution.

having done the basics we could go deeper and think of smart ways to fix the out-of-order writes leveraging the usb/scsi layer without big changes to the firmware base.. if that is even possible :-)

flit, 2019-02-06:
Closing issue. DAPLink is OS-agnostic firmware. If there are any bugs that prevent it from working with FreeBSD or any other OS, then we will track and fix the bugs as separate issues.

closed 2019-02-06 by flit