ARMmbed/DAPLink #315: Test / Win10 / Assert

cederom, 2017-09-11:
On Windows10 machine I get this permanent test failure:
```
SubTest: Build test configuration
(...)
Test configurations to be run:
    0: APP=k20dx_frdmk64f_if BL=k20dx_bl Board=FRDM-K64F Target=FRDM-K64F

Fimrware that will not be tested:
(...)

Board: Name=FRDM-K64F Unique ID=0240000028884e45001070116bf000478021000097969900
Application: Name=k20dx_frdmk64f_if
Bootloader: Name=k20dx_bl
Target: Name=FRDM-K64F
SubTest: load_bootloader
SubTest: set_mode
programming took 2.75500011444 s
SubTest: wait_for_remount
unmount took 0.668999910355 s
mount took 2.77900004387 s
SubTest: set_mode
Bootloader crc: 0xef0dad3a
SubTest: load_interface
SubTest: set_mode
changing mode IF -> BL
SubTest: wait_for_remount
unmount took 1.06900000572 s
mount took 2.99699997902 s
programming took 5.60099983215 s
SubTest: wait_for_remount
unmount took 0.753000020981 s
mount took 2.36400008202 s
SubTest: set_mode
Interface crc: 0x273c9d8e
SubTest: daplink_test
SubTest: Assert Test
Triggering assert by creating ASSERT.ACT
SubTest: wait_for_remount
unmount took 1.52799987793 s
mount took 2.20000004768 s
SubTest: test_fs
chkdsk returned 0
SubTest: test_fs_contents
File D:\MBED.HTM valid
File D:\DETAILS.TXT valid
File D:\ASSERT.TXT valid
Skipping non file item D:\System Volume Information
SubTest: test_details_txt
SubTest: parse_kvp_file
SubTest: test_details_txt
SubTest: parse_kvp_file
Checking that assert file was created
Checking that assert file persists if->bl
changing mode IF -> BL
unmount took 0.799999952316 s
mount took 3.08600020409 s
chkdsk returned 0
File D:\HELP_FAQ.HTM valid
File D:\DETAILS.TXT valid
File D:\ASSERT.TXT valid
Checking that assert file persists bl->if
changing mode BL -> IF
Traceback (most recent call last):
  File "../test/run_test.py", line 701, in <module>
    main()
  File "../test/run_test.py", line 679, in main
    tester.run_tests()
  File "../test/run_test.py", line 238, in run_tests
    daplink_test(test_configuration, test_info)
  File "C:\(...)\DAPLink.git\test\test_daplink.py", line 138, in daplink_test
    test_assert(workspace, test_info)
  File "C:\(...)\DAPLink.git\test\test_daplink.py", line 223, in test_assert
    board.set_mode(board.MODE_IF)
  File "C:\(...)\DAPLink.git\test\daplink_board.py", line 310, in set_mode
    with open(start_if_path, 'wb') as _: pass
IOError: [Errno 22] invalid mode ('wb') or filename: 'D:\\START_IF.ACT'
DAPLINK TESTING ERROR, TEST 1505153767 FAILED MISERABLY WITH STATUS 1, EJECTING!
./test_loop.sh: line 23: say: command not found
(venv-win-27)

$ mbedls.exe
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | D:          | COM4        | 0240000028884e45001070116bf000478021000097969900 | 0243            |
| LPC812        | LPC812[0]            | F:          | COM5        | 10500000e062b0bf00000000000000000000000097969902 | 0243            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
```

cederom, 2017-09-21:
This seems to be **BLOCKER** related with `chkdsk` on win10 blocking write access to a `DAPLINK` drive.

cederom, 2017-09-22:
CHKDSK on win10 causes test framework to fail. It marks fs as dirty, so fs gets read-only. Disabling test_fs() allows test to follow. Probably will have to check VFS implementation and fix to support CHKDISK/FSCK operations..

cederom, 2017-09-22:
Simply disabling CHKDSK does not fix the issue. Other test fail with strange situations occuring..

brianesquilona, 2018-12-28:
Cannot reproduce with current test environment.

closed 2018-12-28 by brianesquilona