ARMmbed/DAPLink #316: NRF5284_DK instability and issue with firmwares

dhwalters423, 2017-09-15:
I found several issues working with NRF5284_DK.

1) The mbed board page for the nRF52-DK links to firmware version 0218 of CMSIS-DAP. When running 'mbed detect' using this firmware, the platform_name reports as a nRF51x board, and this is outdated.

2) The Nordic website for the board: https://www.nordicsemi.com/eng/Products/nRF52840-Preview-DK/(language)/eng-GB lists mbed interface version 0221. This should be updated to 0243.

3) There doesn't seem to be a release of 0243 for the NRF5284_DK.

4) When flashed with the version 0221 from the Nordic website, 'mbed detect' and 'mbed ls' report the correct platform_name, but this is extremely unstable. Examples include mbed-os-example-blinky.hex causing an error that there is not enough space on the board, followed by a crash. Even when the programming of the board appears to be successful (no crash), there is no blink on the board. I've also found this in the mbed-os-example-ble examples, that either crash on programming the board, or fail to run.

5) The version of J-Link interface found on the Nordic site for the board: https://www.nordicsemi.com/eng/Products/nRF52840-Preview-DK/(language)/eng-GB seems to be more stable (less crashes), but even when you program the board with blinky, blinky fails to run.

6) When programmed with the above mentioned J-Link interface, the output of ``mbedls`` and `mbed detect` report the wrong platform name. Example:
```

C:\Users\davwal04\Desktop\Out of box campaigns\mbed OS 5.6\mbed-os-example-blinky>mbedls
+---------------+----------------------+-------------+-------------+--------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id    | daplink_version |
+---------------+----------------------+-------------+-------------+--------------+-----------------+
| LPC2368       | LPC2368[0]           | D:          | COM11       | 000683907488 | unknown         |
+---------------+----------------------+-------------+-------------+--------------+-----------------+
```

Found with Windows 10, running mbed os 5.6 during the Out of Box campaign.

cederom, 2017-09-15:
Hello @dhwalters423 :-) I also have this kit and I am eager to work on this subject.. as for now we are focusing on a stable release, after that we can get into help with nRF52840-PDK to Nordic (I have updated label of the issue accordingly)..

Regarding your questions:
1. Please note that mbed `nRF52-DK` is `nRF52832-DK` not `nRF52840-PDK`..
2. We have no control over external sites, but we will update that information after we make a release that supports this kit.
3. It's not yet supported, sorry, but I am eager to help make it work! Nordic is working on that already. They have full support on our side, let them know if you can, we are here to help :-)
4. Please note that `mbed-ls` (https://github.com/ARMmbed/mbed-ls) is an external project to DAPLink..
5/6. Please note that this is still a **PREVIEW DEVELOPMENT KIT** under development, even its hardware changes, but we are open to help making it work witm ARM mbed! :-)

dhwalters423, 2017-09-15:
Thanks @cederom for the response :)

1) Understood. It was the closest platform I could see to the 52840 in the Mbed supported platforms.

2 - 3) I'll wait until we have full support for the 52840 and ask them to update to the newest DAPLink when it is supported. I'll also work on notifying them that 0221 is very unstable on the platform.

4-5-6)  My gut instinct was that the mbed-ls tools failing to report had something to do with the interface version since each interface (J Link, DL 0221, DL 0218) all reported different platform names (nRF51xxx, nRF52840, LPC2368). Shot in the dark, but could this have to do with the details.txt  that is bundled with each interface version? At least, from my understanding of mbed-os, it grabs this information from details.txt.

adbridge, 2017-09-18:
@c1728p9

pan-, 2017-09-22:
> 4-5-6) My gut instinct was that the mbed-ls tools failing to report had something to do with the interface version since each interface (J Link, DL 0221, DL 0218) all reported different platform names (nRF51xxx, nRF52840, LPC2368). 

@dhwalters423 Correct, there is an ID which has been reserved for the kit and it is present in mbed-ls `manufacture_ids` ([here](https://github.com/ARMmbed/mbed-ls/blob/3dc899e5dd2dd9f064918076c9b3bae15d06cc50/mbed_lstools/lstools_base.py#L205)) . However if the interface firmware returns a different ID then mbed-ls cannot identify correctly the target.

dhwalters423, 2017-09-22:
What would be a way forward here? Is this something Nordic can fix  as it seems DAPLink is incompatible with the board at the moment.

cederom, 2017-09-22:
Should we ask on mbed / mbed-ls to add this board / target if its not already there?

dhwalters423, 2017-09-22:
@cederom

cederom, 2017-09-22:
@dhwalters423 we will have a next release soon, about 2 weeks, after that I will get into PDK :-)

dhwalters423, 2017-10-05:
I just did some further testing and found that the .bin `0243_sam3u2c_mkit_dk_dongle_nrf5x_0x5000.bin` included in the 0243 release is actually working okay and seems to be stable. There is still a defect with mbed-ls however:

```
$ mbed detect

[mbed] Detected NRF51822, port COM18, mounted D:, DAPLink version 0243
[mbed] Supported toolchains for NRF51822
+--------+-----------+-----------+-----+---------+-----+
| Target | mbed OS 2 | mbed OS 5 | ARM | GCC_ARM | IAR |
+--------+-----------+-----------+-----+---------+-----+
+--------+-----------+-----------+-----+---------+-----+
Supported targets: 0
```

vergil-SI, 2017-11-25:
Has there been any progress on this issue that I've missed by chance?

MarceloSalazar, 2017-12-05:
What's the status of this?

c1728p9, 2017-12-05:
The NRF52840_PDK is still not a supported target in DAPLink. Until support has been added, ```0243_sam3u2c_mkit_dk_dongle_nrf5x_0x5000.bin``` should not be used with the NRF52840_PDK.

bridadan, 2018-03-19:
FYI mbed-ls now supports JLink firmware and will correctly report the NRF52_DK board. However I don't have a NRF52840_DK to test with. If anyone here as a NRF52840_DK with JLink loaded on it, all I need is a .zip of the board's MSD files. Feel free to raise an issue on mbed-ls to get the target properly supported: https://github.com/ARMmbed/mbed-ls

closed 2018-12-28 by brianesquilona