ARMmbed/DAPLink #322: The latest LPC11U35 firmware update failed

neomissing, 2017-10-05:
Compile the latest source code, generate LPC11U35 firmware and found a direct U disk update firmware will fail, the previous firmware no problem, but the latest firmware can be used to burn the simulator can work, where changes lead to this problem?

c1728p9, 2017-10-05:
Hi @neomissing, what do you mean by U disk update? What file are you loading? Is the device in CRP disabled mode?

neomissing, 2017-10-05:
lpc11u35_lpc824xpresso_if,the device in CRP disabled mode,update LPC11U35 firmware,

c1728p9, 2017-10-05:
Try using lpc11u35_lpc824xpresso_if_crc.bin instead. CRP disabled mode requires the CRCed version of the binary.

neomissing, 2017-10-05:
Thanks,I try lpc11u35_lpc824xpresso_if_crc.bin ,it's OK.

c1728p9, 2017-10-05:
Good to hear. Can this issue be closed?

closed 2017-10-05 by neomissing