ARMmbed/DAPLink #323: NRF5x BOARD IDs are incorrect except for nRF51822

screamerbg, 2017-10-06:
I'm using the unified nRF5x interface firmware in release 0243 on both nRF52840 and nRF52832, and I'm getting a BOARD ID for nRF51822, which is incorrect.

Here are the IDs in the Mbed Web/LS database:

| TARGET | BOARD ID | Slug | Vendor |
| -- | -- | -- | -- |
| nrf52840_dk | 1102 | Nordic-nRF52840-DK | Nordic |
| nrf52_dk | 1101 | Nordic-nRF52-DK | Nordic |
| nrf51_dongle | 1120 | Nordic-nRF51-Dongle | Nordic  |
| nrf51_dk | 1100 | Nordic-nRF51-DK | Nordic |
| nrf51822 | 1070 | Nordic-nRF51822 | Nordic |

Perhaps separate releases can be provided for these where the BOARD ID is hardcoded. Not sure whether a unified firmware can detect the difference between nRF51 DK and nRF51 Dongle for example as these are the same interface chips and MCUs, just a different form factor.

cc @c1728p9

dhwalters423, 2017-10-06:
Related to the mbedls oddities as reported in #316

c1728p9, 2017-10-06:
Hi @screamerbg currently the nrf52840 isn't a supported target, which is why the board ID is showing up incorrectly, and why programming doesn't work. The project sam3u2c_mkit_dk_dongle_nrf5x_if builds firmware for the nrf52_dk, nrf51_dongle, nrf51_dk and nrf51822, but not for the nrf52840_dk. Supported for multiple boards running from the same image was added at the request of @gamnes. To get support for the nrf52840_dk Nordic should add this to DAPLink.

closed 2018-12-28 by brianesquilona