ARMmbed/DAPLink #333: With 0244 the OpenSDA refuses to talk to other targets

therealprof, 2017-10-29:
Today I've upgraded my OpenSDA on the FRDM-KL26Z from 0242 to 0244 and ever since then I cannot properly use it with OpenOCD in isolated mode to program an ATSAMD20 anymore which worked perfectly fine previously.

The observed changes in behaviour are that the target is now held in reset upon reconnection, power up of the debugger or power up of the target but refuses to be detected:
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
none separate
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 1.0
Info : SWCLK/TCK = 0 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 400 kHz
in procedure 'program'
in procedure 'init' called at file "embedded:startup.tcl", line 495
in procedure 'ocd_bouncer'
** OpenOCD init failed **
shutdown command invoked
```

The only way I've found to reconnect the target (and also get it out of reset) is to use J8 to reconnect SWCLK to the KL26Z and rerun OpenOCD:
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
none separate
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 1.0
Info : SWCLK/TCK = 0 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 400 kHz
Info : SWD DPIDR 0x0bc11477
Info : SWD DPIDR 0x0bc11477
Error: Failed to write memory at 0xe000200c
Info : at91samd20e15.cpu: hardware has 2 breakpoints, 2 watchpoints
Info : SWD DPIDR 0x0bc11477
Error: Failed to write memory and, additionally, failed to find out where
in procedure 'program'
in procedure 'reset' called at file "embedded:startup.tcl", line 500
in procedure 'ocd_bouncer'
in procedure 'ocd_process_reset'
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 248
in procedure 'at91samd20e15.cpu' called at file "embedded:startup.tcl", line 331
in procedure 'ocd_bouncer'

Info : SWD DPIDR 0x0bc11477
TARGET: at91samd20e15.cpu - Not halted
in procedure 'program'
in procedure 'reset' called at file "embedded:startup.tcl", line 500
in procedure 'ocd_bouncer'

Info : SWD DPIDR 0x0bc11477
Polling target at91samd20e15.cpu failed, trying to reexamine
Info : SWD DPIDR 0x0bc11477
Info : SWD DPIDR 0x0bc11477
Info : SWD DPIDR 0x0bc11477
Info : SWD DPIDR 0x0bc11477
Info : at91samd20e15.cpu: hardware has 2 breakpoints, 2 watchpoints
** Unable to reset target **
Polling target at91samd20e15.cpu failed, trying to reexamine
Info : SWD DPIDR 0x0bc11477
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 100ms
shutdown command invoked
```

This will also let my ATSAMD20 out of reset. Then disconnect J8 again and now it works until I restart or accidentally hit the reset button:
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
none separate
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 1.0
Info : SWCLK/TCK = 0 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 400 kHz
Info : SWD DPIDR 0x0bc11477
Info : at91samd20e15.cpu: hardware has 4 breakpoints, 2 watchpoints
target halted due to debug-request, current mode: Thread
xPSR: 0x61000000 pc: 0x000000a0 msp: 0x20001000
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD20E15A (32KB Flash, 4KB RAM)
wrote 2048 bytes from file target/thumbv6m-none-eabi/release/examples/fading_glamour in 0.391204s (5.112 KiB/s)
** Programming Finished **
** Verify Started **
verified 1020 bytes in 0.162986s (6.112 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

cederom, 2017-10-29:
Hello @therealprof, have you tried with pyOCD?

therealprof, 2017-10-29:
I have not, but I might.

I'm not quite sure the relationship of using pyOCD with this obvious regression.

c1728p9, 2017-10-30:
Thanks for the report @therealprof. I have a couple of questions to help narrow this down.
* If you revert DAPLink back to 0242 are you able to program again?
* If so, does 0243 work?
* What was the last firmware you loaded on the KL26?

I have seen a similar problem to the one you described, but in my case, it was due to bad firmware loaded on the target (KL26). Because of the bad firmware the target was resetting constantly and since the reset line of the target is shared with the externally connected device, it was getting reset as well. **The solution in my case was to flash firmware good firmware to the on board target so it wasn't resetting.**

therealprof, 2017-10-30:
Well, with the "workaround" I am able to program. I just flashed back to 0242 again and it still does not work directly anymore. The other thing I did earlier was to upgrade the bootloader to 0244, maybe that is the real culprit?

c1728p9, 2017-10-30:
Hi @therealprof updating the bootloader won't change the behavior of 0242 or 0244 so this isn't the problem. Since 0242 isn't working either, its likely the firmware that is on the KL26 that is causing problems. 
Have you tried re-flashing the KL26 with a good image?

therealprof, 2017-10-30:
@c1728p9 Uhm, I can certainly try to flash "a good image" to the KL26 but I haven't really touched the KL26 at all, I only bought the board so I have access to a (3V3) DAPlink debugger which I need for the "cold plug" support to get that darn ATMSAMD20 out of a locked state which I happen to produce quite a lot.

therealprof, 2017-10-30:
@c1728p9 Oh wow, you were right. The firmware on the KL26 (probably the bootloader upgrade by accident) was interfering with the debugger. Thanks for the support.

closed 2017-10-30 by therealprof

c1728p9, 2017-10-30:
I'm glad I could help :smile: