ARMmbed/DAPLink #338: Core / Unify HW and SW reset sequences

cederom, 2017-11-02:
This also relates to SWD Access, it will be reworked after https://github.com/ARMmbed/DAPLink/issues/337..

cederom, 2017-12-18:
The solution is proposed here: https://github.com/ARMmbed/DAPLink/pull/371

loverdeg-ep, 2019-05-22:
@0xc0170 @linlingao   Letting you know I just accidentally realized I have the ability to modify the project cards contained within with the ARMmbed group.

linlingao, 2019-05-22:
@loverdeg-ep Thanks for letting us know. I have no idea how Project cards are being used.

0xc0170, 2019-05-23:
> @0xc0170 @linlingao Letting you know I just accidentally realized I have the ability to modify the project cards contained within with the ARMmbed group.

Correct, currently set to any member of our group.