ARMmbed/DAPLink #341: LPC4322 USB is unstable

c1728p9, 2017-11-02:
The lpc4322's USB driver is unstable and does not pass testing. This needs to be fixed before this HIC can be tested in CI.

c1728p9, 2017-11-02:
@mmahadevan108 can you make these updates since you are maintaining the lpc4322?

mmahadevan108, 2017-11-29:
@c1728p9 Can you provide more details on how to reproduce this failure.

c1728p9, 2017-11-29:
If you run the DAPLink test suite on linux then this failure will occur.

acassis, 2020-06-28:
I'm also facing issues with DAPLink firmware on LPC4322 of the iMXRT1060_EVK board. For small files (less than 128KB) sometimes it works, i.e. I was able to copy this file after many attempts:
$ ls -l nuttx.bin
-rwxrwxr-x 1 alan alan 73436 Jun 28 14:32 nuttx.bin

For big files (>128KB) it never worked.