ARMmbed/DAPLink #354: FAIL.TXT : Transfer timeout

akrnvm, 2017-11-23:
Hi,
can anybody help me to resolve the daplink issue.
my daplink is not able to flash bins to target.
in DAPLINK, when i place a new bin file, its comes with FAIL.TXT with transfer timeout message.

please help me to create traget.c file and flash_blob.c.

Device info is given below.
```
struct FlashDevice const FlashDevice  =  {
   FLASH_DRV_VERS,             // Driver Version, do not modify!
   ONCHIP,                     // Device Type
   0x00000000,                 // Device Start Address
   0x00180000,                 // Device Size (1536kB)
   4096,                        // Programming Page Size
   0,                          // Reserved, must be 0
   0xFF,                       // Initial Content of Erased Memory
   100,                        // Program Page Timeout 100 mSec
   5000,                       // Erase Sector Timeout 5000 mSec

// Specify Size and Address of Sectors
   0x001000, 0x000000,         // Sector Size   4kB (384 Sector) 
   SECTOR_END
```

```
 <memory id="IROM1"                           start="0x00000000" size="0x00180000" startup="1" default="1"/>
          <memory id="IRAM1"                           start="0x20000000" size="0x00030000" init   ="0" default="1"/>
          <algorithm name="Flash/XXXXx_code_1536.FLM" start="0x00000000" size="0x00180000"  RAMstart="0x20000000"  RAMsize="0x2000"  default="1"/>
          <algorithm name="Flash/XXXXx_data_32.FLM"  start="0x30000000" size="0x00008000"  RAMstart="0x20000000"   RAMsize="0x2000"  default="1"/>
 <memory id="IROM1"                           start="0x00000000" size="0x00180000" startup="1" default="1"/>
          <memory id="IRAM1"                           start="0x20000000" size="0x00030000" init   ="0" default="1"/>
          <algorithm name="Flash/XXXXx_code_1536.FLM" start="0x00000000" size="0x00180000"  RAMstart="0x20000000"  RAMsize="0x2000"  default="1"/>
          <algorithm name="Flash/XXXXx_data_32.FLM"  start="0x30000000" size="0x00008000"  RAMstart="0x20000000"   RAMsize="0x2000"  default="1"/>
```

is the below traget.c file is propper ?
```
target_cfg_t target_device = {
    .sector_size    = KB(32),
    .sector_cnt     = (KB(1536) / KB(32)),
    .flash_start    = 0,
    .flash_end      = KB(1536),
    .ram_start      = 0x20000000,
    .ram_end        = 0x20004000,
    .flash_algo     = (program_target_t *)&flash,
    .erase_reset    = 1
};

```
flash algo generated by tools provided by mbed and its working with other controllers.
same method used for this controller and its not working.
Please confirm below flash algorithm for target is proper 
```
static const program_target_t flash = {
    0x20000021, // Init
    0x20000037, // UnInit
    0x2000003f, // EraseChip
    0x200000d5, // EraseSector
    0x200001b7, // ProgramPage

    // BKPT : start of blob + 1
    // RSB  : blob start + header + rw data offset
    // RSP  : stack pointer
    {
        0x20000001,
        0x200002d8,
        0x20000800
    },

    0x20000000 + 0x00000A00,  // mem buffer location
    0x20000000,               // location to write prog_blob in target RAM
    sizeof(flash_prog_blob),   // prog_blob size
    flash_prog_blob,           // address of prog_blob
    0x00001000       // ram_to_flash_bytes_to_be_written
};
```

cederom, 2017-11-23:
Hey @akrnvm, what Target are you referring to? What Target does work and does not with current DAPLink? Are you sure Bootloader and Interface Firmwares matches your HIC and Layout (i.e. HIC is SAM3U and Layout is `0x5000`)? Do you use `0244` release or custom build?

akrnvm, 2017-11-23:
im using 0244 release.
Interface firmware is flashing to my Daplink without any error and able to flash via KEIL IDE.
flashing  application only facing problem.
working Target flash alg info:
```
struct FlashDevice const FlashDevice  =  {
   FLASH_DRV_VERS,             // Driver Version, do not modify!
   ONCHIP,                     // Device Type
   0x00000000,                 // Device Start Address
   0x00020000,                 // Device Size (128kB)
   256,                        // Programming Page Size
   0,                          // Reserved, must be 0
   0xFF,                       // Initial Content of Erased Memory
   100,                        // Program Page Timeout 100 mSec
   3000,                       // Erase Sector Timeout 3000 mSec

// Specify Size and Address of Sectors
   0x001000, 0x000000,         // Sector Size   4kB ( 8 Sector) 
   0x008000, 0x008000,         // Sector Size  32kB ( 3 Sector)
   SECTOR_END
};

```

```
static const program_target_t flash = {
    0x20000021, // Init
    0x20000035, // UnInit
    0x2000003d, // EraseChip
    0x200000B1, // EraseSector
    0x20000129, // ProgramPage

    // BKPT : start of blob + 1
    // RSB  : blob start + header + rw data offset
    // RSP  : stack pointer
    {
        0x20000000 + 0x00000001,
        0x200001F0,
        0x20001000 - 0x20
    },
    0x20000000 + 0x00000200,   // mem buffer location
    0x20000000,                // location to write prog_blob in target RAM
    sizeof(flah_blob),      // prog_blob size
    flsah_blob,              // address of prog_blob
    0x00000100                 // program_buffer_size, largest size that can be written in a single call to program page
};
```

procedure is used for both are same.

my Queries?
1. is the target.c and flash_blob.c files only depend to target board.
2. algorithm RAM_SIZE is 0x2000 in KEIL IDE, is same size require in algorithm.

cederom, 2017-11-23:
Okay, so you have problem with flashing your Target.. if the target is not yet supported you need to create a dedicated flash algo for the device, each device (family) has its own dedicated flash algo.. everything needs to exactly match your Target hardware.

Take a look at documentation. PORTING section may be helpful for you, or the whole handbook [1] if you want to know more. If you just need info on flashing, take a look at ARM mbed FlashAlgo project [2] maybe also Keil uVision Flash Algorithms [3].

We could help you out but we need to know and have the Target at hand. Note the `test/run_test.py` can help you out test your implementation in an automatted manner, see TESTING [1] for more information.

[1] https://github.com/cederom/DAPLink/blob/cederom_doc/doc/DAPLinkHandbook.md
[2] https://github.com/mbedmicro/FlashAlgo
[3] http://www.keil.com/support/man/docs/uv4/uv4_fl_flash_algorithms.htm

akrnvm, 2017-11-24:
hi,
Thank you for the information.
In my DAPLINK code auto increment page size is 1024. 
`#define TARGET_AUTO_INCREMENT_PAGE_SIZE    (1024)`
In Target flash algo programing page size 4096.
`4096,                        // Programming Page Size`
is this cause any issue?

c1728p9, 2017-11-27:
Hi @akrnvm, from the error you are getting, it sounds like your device isn't even starting the flashing process. The think I would check is that the **binary file you are loading has a valid vector table**, as validated here:
https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/validation.c#L31-L58

Another thing you could try is loading a hex file with the same contents. Hex detection is different from binary so this would tell you if it is a problem detecting binary files.

akrnvm, 2017-11-28:
thanks for the information.
I have tried with both .bin and .hex files.

when in using software reset this error message shows "timeout" and in hardware reset error message is coming as
"The interface firmware FAILED to download the flash programming algorithms to the target MCU".

is it related to target reset timing issue.

c1728p9, 2017-11-28:
Hi @akrnvm I noticed you closed this issue. Were you able to fix the problems?

cederom, 2017-11-28:
Hello @akrnvm @c1728p9 I am working on this particular reset stuff right now! Any feedback and problems / solutions are welcome here and now! I can make the fixes :-)

akrnvm, 2017-12-04:
my issue got resolved by changing 
1. Target reset method from sw_reset to hw_reset.
2. Flash algorithm  program_buffer_size 0x00001000 (generated by  flash algo tool) to  0x00000200
 ` 0x00000200                 // program_buffer_size, largest size that can be written in a single call to program page`