ARMmbed/DAPLink #357: interface / dap / swd_init() improvements

cederom, 2017-11-29:
```
uint8_t swd_init(void)
{
    //TODO - DAP_Setup puts GPIO pins in a hi-z state which can
    //       cause problems on re-init.  This needs to be investigated
    //       and fixed.
    DAP_Setup();
    PORT_SWD_SETUP();
    return 1;
}
```
This function is called several times, during the run, for instance as Target Reset mechanism.. maybe we could use simple flag that marks port already initialized so no need to reinitialize.. or use function that could optionally force port reinitialization? :-)

flit, 2017-11-30:
I've had similar thoughts. Seems silly to completely reinit the port and everything multiple times, when all we want to do is ensure that the port _was_ inited previously.

Also, take a look at #189. While not directly related, it's in the same init/shutdown functional area of swd_host.

cederom, 2017-11-30:
Thanks @flit :-) As for now I am focusing on RESET sequences, but noted this TODO as next goal :-) Thanks for the important reference to https://github.com/ARMmbed/DAPLink/pull/189 :-)