ARMmbed/DAPLink #366: DAPLink boot sequence problems on K20 interface

c1728p9, 2017-12-12:
From the blog post here there appears to be a reset sequence problem effecting some NXP boards based on the K20:
https://mcuoneclipse.com/2017/10/29/adding-a-delay-to-the-arm-daplink-bootloader/

For reference the code to delay and allow the voltage rails to stabilize can be found here:
https://github.com/ARMmbed/DAPLink/blob/master/source/hic_hal/freescale/k20dx/gpio.c#L67-L73

c1728p9, 2017-12-12:
@flit @mmahadevan108 @ErichStyger the code in DAPLink can be easily changed to increase the boot delay. How long of a delay is needed to reliably let the rails drain? I did some profiling locally a while back and tried to set this with a fairly large margin, but this doesn't appear to be enough on all the configurations.

CC: @maclobdell

ErichStyger, 2017-12-12:
I had to use a delay up to 500-1000 ms, especially for off-board targets where the power-up sequence is much more disconnected.

brianesquilona, 2018-12-28:
I think this issue is resolved, anyone can reopen it if it's not.

closed 2018-12-28 by brianesquilona