ARMmbed/DAPLink #373: mbed-HDK STM32 HIC not supported?

davidrea, 2017-12-20:
The mbed-HDK repository contains [a "bare design project"](https://github.com/ARMmbed/mbed-HDK/tree/master/Bare%20Design%20Projects/ARM-mbed/INTERFACE%20DAPLink/STM32F103CBT6) for a debug probe using the STM32F103 as a HIC. There does not, however, appear to be a [DAPLink build](https://github.com/ARMmbed/DAPLink/releases) or for this target. I also don't see it appear in [boards](https://github.com/ARMmbed/DAPLink/tree/master/source/board), [targets](https://github.com/ARMmbed/DAPLink/tree/master/source/target) or [hic_hal](https://github.com/ARMmbed/DAPLink/tree/master/source/hic_hal)s. I also don't see any boards using the 'F103 included on the [firmware release site](https://armmbed.github.io/DAPLink/)

Is the STM32F103CBT - which appears to remain an official HDK-recommended HIC - still supported by DAPLink? Am I looking in the wrong place(s) for official builds?

I am aware of the x893 fork of CMSIS-DAP that appears to offer a [build](https://github.com/x893/CMSIS-DAP/tree/master/Firmware/STM32) compatible with the circuit. An official build from mainline, up-to-date DAPLink would be much preferred.

cederom, 2017-12-24:
Hello @davidrea, replacing ST-LINK is a bit problematic at the moment but possible, you would have to manually overwrite the ST-LINK firmware on STM32F103 HIC.. unfortunately there is no official DAPLink build for that HIC because it uses proprietary bootloader that uses encryption we don't have access to. You can build the bootloader and interface firmware and then use SWD/JTAG to flash it into HIC..

We have discussed this issue recently with @sg- and @c1728p9 maybe we could provide the sources / builds if there was enough interest..

Krakonos, 2018-10-29:
Hi!

Has there been any work on it? I'm currently confused by this as well. I know replacing ST-LINK is a bit hard to do, but there seem to be other probe schematics and boards in the mbed-HDK repository. For example, I have found this STM32-DIPDAP project:

https://github.com/ARMmbed/mbed-HDK-Eagle-Projects/tree/master/DAPLink/Probes/STM32-DIPDAP/v1.0.0

But there seem to be no firmware build (and currently I don't have Keil license available to build it by myself, nor do I want to do it).

My goal is to include a programmer with UART convertor on a development board I'm designing. Though I could do it with the LPCxxx projects, I do prefer STM32s as I already have experience with them and do have the tooling available (LPC would probably be no different, but I still prefer STM32s just to be sure).