ARMmbed/DAPLink #375: Documentation is unclear

drewcassidy, 2018-01-10:
I wanted to compile a version of this for an lpc11u35 CMSIS-DAP programmer for use with the nRF51822, but honestly cannot figure out the documentation here on how to compile it, or if target-specific versions are even necessary (I asume thats what the model numbers after the dash mean?). 

- the [docs](https://github.com/ARMmbed/DAPLink/blob/master/docs/DEVELOPERS-GUIDE.md) mention a .bat file... but I'm not on windows
- the 'setup' instructions mention a series of commands involving a python tool, but it must have updated at some point and no longer produces any scripts, or I have a missing prerequisite

am I missing something, or is this repo in desperate need of better documentation?

cederom, 2018-01-11:
Hey @drewcassidy, I understand your concerns. At the moment DAPLink can be compiled only with Keil uVision.. from the sources you need to generate uVision projects using `progen` utility (part of `requirements.txt` to your system or virtualenv), then you can simply open given project and build it with `F7` hotkey.. projects are generated based on `projects.yaml` and produces all supported HIC+Target variants.

drewcassidy, 2018-01-11:
will this compile in the evaluation version of uVision? or do I need a license?

cederom, 2018-01-11:
The evaluation version can produce only 32KB of code [1], so you will need a license sorry.. this was the only tool that could produce reliable quality code.. but we plan to port DAPLink to use ARM mbed this year so you could use ARM mbed Online Compiler or local GCC/CLANG..

[1] http://www.keil.com/demo/limits.asp

cederom, 2018-01-11:
What is your goal @drewcassidy? Are release binaries [1] not enough for you? There is another possibility we could use - add pull request here on GitHub and our CI backend will build it for you then run the automated tests on the build - you can get the binaries and verification using our license and resources..

[1] https://github.com/ARMmbed/DAPLink/releases

drewcassidy, 2018-01-11:
I was hoping to compile a version of it for [this board](https://github.com/OSHChip/OSHChip_CMSIS_DAP_V1.0_Docs) and also possibly tweak the firmware a bit

cederom, 2018-01-11:
Most nRF boards use SAM3U2 as the HIC (chip for DAPLink). On the provided board you have 11U35 which is known HIC to work with Freescale Targets. For the moment Target support is hardcoded as flash-algo into firmware for UMS (drag-n-drop flashing) operations. Still if the pinouts match (and the design seems to be copy-pasted) you can try to use existing release based on 11U35 HIC to get HID pyOCD debug and CDC virtual com port. This would give you access to UART and DEBUG which seems to be enough for development as you can use DEBUG also for Target Flashing..

If you want to create 11U35 + nRF51 variant feel free to add new board and we can leverage CI backend for build and testing :-)

wzt19910506, 2018-01-13:
I use lpc11u35 as HIC to burn nrf51 successfully, but there is still a problem. Every time after burning through drag and drop nrf51822 does not automatically reset. By comparing with the logic analyzer and microbit. In the end the burning swdio pin above the lack of a 20ms left and right low level, and it is this low level you can reset the nrf51822. now in confusion to the source code do not know where to make lpc11u35 also allows nrf51822 to achieve automatic reset after burn

cederom, 2018-01-14:
Hey @wzt19910506, as mentioned above Drag-n-Drop flashing only works for a custom dedicated HIC+Target firmware build. Generic firmware / HIC should give you SWD Debug and Virtual COM Port UART.

Reset for nRF51 is known to be quirky using software reset and based on the GPIO settings on the nRF-DK which uses SAM3U3 HIC. But we are working on new RESET Unification that should work on any board you can try that from and report back https://github.com/ARMmbed/DAPLink/pull/371. Its almost done and soon will become part of master so 11U35 HIC should also benefit working reset on nRF boards.. please let me know if that works for you.

wzt19910506, 2018-01-16:
@cederom yes，thank you very much! I will try it.

drewcassidy, 2018-01-16:
I'm not even a fan of the MSD programming (the eject warnings get annoying) so thats fine. I was going to try disabling it entirely, honestly

wzt19910506, 2018-01-16:
@drewcassidy It has its advantages, it can complete the flashing without any PC software, don't you think it's very convenient?

drewcassidy, 2018-01-16:
sure, I just dont like how after each time you program it you get yet another eject warning