ARMmbed/DAPLink #379: RFE : add support for "IBDAP Adapter from Armstart"

captcha1, 2018-01-17:
Please add support for IBDAP (sold on Tindie & Adafruit : https://www.adafruit.com/product/2764 )

It's based on LPC11U35 . It seems to have JTAG / SWD , SWO , UART , etc.

Schematic is here : https://halckemy.s3.amazonaws.com/uploads/pdf_file/file/55500/IBDAP-LPC11U35_SCH.pdf

Some background info : https://www.hackster.io/armstart/ibdap-affordable-cmsis-dap-jtag-swd-debug-probe-e6d9a4

flit, 2019-02-14:
Agreed we need to add support for IBDAP. The original firmware doesn't seem to be available anymore except as source on GitHub, and that doesn't even build successfully (for me, at least).

This board has full connections for JTAG and SWO, which is unique for LPC11U35-based probes.

We also need to work with Adafruit and Seeed to link to the DAPLink-based firmware when its available.

sur5r, 2020-05-15:
Please see devanlai/IBDAP-CMSIS-DAP#1 (or sur5r/IBDAP-CMSIS-DAP@74da54a). This builds for me on using arm-none-eabi-gcc 8.3.1.

flit, 2020-05-18:
We'd be happy to accept a pull request adding IBDAP support.

sur5r, 2020-05-19:
I know. I just wanted to point out that the original firmware is still usable, maybe as starting point for DAPLink support. I was not yet able to compile DAPLink myself, which stopped me from finding out what's needed to support this board.

ibdap, 2020-11-25:
The origin IBDAP source code is now relocated to: https://github.com/ibdap/ibdap-cmsis-dap
From what we have investigated, DAPLink needs some work to replace proprietary dependencies in order to support building the source code from gcc. Am I right @flit ?

mathias-arm, 2021-05-27:
We have been working on GCC Support. You might want to try the `lpc11u35_if` (or `lpc11u35_tiny_if` which has a different PIN mapping) from https://github.com/ARMmbed/DAPLink/wiki/compilers-test-status