ARMmbed/DAPLink #383: got error when running 'progen generate -t uvision'

zhangzq71, 2018-01-26:
I follow the instruction to setup the environment, but got error when running 'progen generate -t uvision', here is the message I got, what is the wrong?

progen generate -t uvision
Traceback (most recent call last):
  File "c:\users\zhang\appdata\local\programs\python\python35\Lib\runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\zhang\appdata\local\programs\python\python35\Lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "D:\EDA\CMISIS-DAP\DAPLink\venv\Scripts\progen.exe\__main__.py", line 9, in <module>
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\project_generator\main.py", line 62, in main
    return args.func(args)
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\project_generator\commands\generate.py", line 30, in run
    if project.generate(args.tool, copied=args.copy, copy=args.copy) == -1:
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\project_generator\project.py", line 552, in generate
    files = exporter(self.project['export'], self.settings).export_project()
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\project_generator\tools\uvision.py", line 487, in export_project
    path, files = self._export_single_project('uvision') #todo: uvision will switch to uv4
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\project_generator\tools\uvision.py", line 466, in _export_single_project
    uvoptx_dic = xmltodict.parse(open(self.uvoptx_file))
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\xmltodict.py", line 328, in parse
    parser.ParseFile(xml_input)
TypeError: read() did not return a bytes object (type=str)

(venv) D:\EDA\CMISIS-DAP\DAPLink>progen generate -t uvision
Traceback (most recent call last):
  File "c:\users\zhang\appdata\local\programs\python\python35\Lib\runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\zhang\appdata\local\programs\python\python35\Lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "D:\EDA\CMISIS-DAP\DAPLink\venv\Scripts\progen.exe\__main__.py", line 9, in <module>
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\project_generator\main.py", line 62, in main
    return args.func(args)
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\project_generator\commands\generate.py", line 30, in run
    if project.generate(args.tool, copied=args.copy, copy=args.copy) == -1:
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\project_generator\project.py", line 552, in generate
    files = exporter(self.project['export'], self.settings).export_project()
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\project_generator\tools\uvision.py", line 487, in export_project
    path, files = self._export_single_project('uvision') #todo: uvision will switch to uv4
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\project_generator\tools\uvision.py", line 466, in _export_single_project
    uvoptx_dic = xmltodict.parse(open(self.uvoptx_file))
  File "d:\eda\cmisis-dap\daplink\venv\lib\site-packages\xmltodict.py", line 328, in parse
    parser.ParseFile(xml_input)
TypeError: read() did not return a bytes object (type=str)

cederom, 2018-02-05:
Hey @zhangzq71, have you tried one of the below?
* Make sure you are using Python 2 (not 3).
* You can check with `which python` what python interpreter you are using
* Make sure you have activated your virtualenv with `source venv-mac/bin/activate` on Unix or `venv-win\Scripts\activatw.bat` on Windows.
* Try to update the dependencies packages with `pip install -U -r requirements.txt`.

Let me know if that helps :-)

zhangzq71, 2018-02-06:
@cederom , Thank you very much, it works now.

closed 2018-02-06 by zhangzq71