ARMmbed/DAPLink #389: Question about reusing HW on DAPLink

andrewc-arm, 2018-02-02:
Hi,

I have a question about reusing HW on DAPLink.
(Since the question is quite generic and can be shared public, I write it here.)

One of our partner has succeeded on making DAPLink HW that enables Maxim MCU Mbed-OS usage. Now he wants to reuse the existing HW and JTAG (SWD) connect to K64F's MCU. It is possible, right? He can reuse the DAPLink HW for any Mbed-Enabled MCU that supports JTAG regardless of Cortex-M3/M4, right?

Thanks,

cederom, 2018-02-02:
Hello @andrewc-arm, yes assuming the physical connections are correct, you should be able to use both HID for CMSIS-DAP Debug and CDC for VirtualCOMPort as a generic utilities.

UMS Drag-n-Drop flashing needs a dedicated Target based Flash-Algo, so each Drag-n-Drop is compiled per Target and creates unique DAPLink firmware, thus, you would need to recompile a firmware for your target if you want to use Drag-n-Drop flashing on a different Target.

There are already generic interfaces such as DIPDAP [1] or SWDAP [2] that you can use the way you mentioned. You can build/use Target specific firmware on them, but still, it is possible to flash a Target using CMSIS-DAP compliant HID channel.

[1] https://os.mbed.com/platforms/DIPDAP-LPC11U35/
[2] https://os.mbed.com/platforms/SWDAP-LPC11U35/

cederom, 2018-02-02:
Also please remember @andrewc-arm to contribute and share back your DAPLink works on Maxim so we can all benefit and play with those chips! :-)

andrewc-arm, 2018-02-07:
Hi, @cederom 
Thanks for the feedback! Now I check it again and it seems it is not based on this repo's code base but using Maxim's DAPLink firmware as it is. Our partner is investigating the options. Now I think we can close the ticket.

closed 2018-02-07 by andrewc-arm

cederom, 2018-02-07:
Try using ANY modern DAPLink HW + FW (latest release is 245) - when connected properly HID Debug and CDC Virtual COM Port should work out of the box with any Cortex-M device. You will not get UMS Drag-n-Drop flashing because it requires custom Target specific Flash Algo.

andrewc-arm, 2018-02-08:
Thanks for the comment.

'UMS Drag-n-Drop flashing' is the key part of uploading the firmware to target MCU, isn't it? Do you know if we can work around and upload the target firmware without Drag-n-Drop? I don't think it is possible, but just want to double check.

cederom, 2018-02-08:
Yes you CAN debug and flash over generic HID CMSIS-DAP channel most Cortex-M and even some Cortex-A devices re-using ANY DAPLink hardware. You can use DAPLink as smart generic SWD/JTAG interface :-)

DAPLink has three functions / endpoints:
1. UMS Drag-n-Drop flashing - provides rapid Target specific flashing with a simple filesystem copy-paste operation. This part is Target specific and needs to be compiled-in for each Target. It is based on a dedicated shellcode binary blob (so called Flash-Algo) that is uploaded over SWD into Target RAM memory and provides extremely fast transfers into Flash memory.
2. HID CMSIS-DAP - provides generic DAP access to Cortex-M and Cortex-A over SWD or JTAG. It allows generic ARM Debug and Memory Access (RAM and Flash). Flashing is not that fast as Flash-Algo because it uses generic SWD operations controlled from a host PC.
3. CDC Virtual COM Port - can help you to communicate with Target UART so you can have commandline interface to your Target Firmware over UART.

UMS (USB Mass Storage) is not the only way to flash. Drag-n-Drop is just a over-simplification one-click of the flashing process :-) It is faster than flashing over SWD. It is easier than flashing with additional tools. Its hardcoded into DAPLink firmware for a given specific Target and this is why DAPLink firmware is provided per-Target. All other parts are fairly generic. Maybe you can find my sketches of DAPLink Handbook handy [1].

For flashing with no UMS but only using HID CMSIS-DAP endpoint you can use Keil uVision or Free and Open pyOCD [2] utility. Feel free to ask if you need more information! :-)

[1] https://github.com/cederom/DAPLink/blob/cederom_doc/doc/DAPLinkHandbook.md
[2] https://github.com/mbedmicro/pyOCD

andrewc-arm, 2018-02-08:
Thanks for the comment. It is really helpful.