ARMmbed/DAPLink #395: Can DAPLink support mk20dx256vlh7?

9fox, 2018-02-18:
Hi,

I have a teensy, and I am just wondering...can DAPLink support this device? mk20dx256vlh7? 

If not, is there some docs/tips/suggestions on migration to this device, please? Thanks very much!

Please let me know if I raised this questions in wrong place. Thanks

Vincent

cederom, 2018-02-18:
Hey @9fox :-) This is the best place to ask questions about DAPLink :-)

MK20 family is supported and widely used as HIC that DAPLink runs on, for instance on most Freescale development boards.

Here are sources:
https://github.com/ARMmbed/DAPLink/tree/master/source/hic_hal/freescale/k20dx

9fox, 2018-02-19:
Hi @cederom thanks very much for your reply! Sounds great to me and I will have a try following your links:-)

Vincent

closed 2018-02-19 by 9fox