ARMmbed/DAPLink #396: JLINK -> DAPLink on nRF52-DK Bootloader update problem

cederom, 2018-02-19:
I have brand new nRF52-DK with JLINK debug firmware. I have problem replacing JLINK with DAPLink on SAM3U HIC. Interface firmware can be updated easily with no problem. But I cannot update the Bootloader, even though I have running DAPLink Interface and Automation is enabled. I was trying using image provided at [1] and [2]. @c1728p9 can you recommend a solution without JTAG/SWD flashing?

What I did on first nRF52-DK:
1. Started JLINK in Bootloader mode and flashed DAPLink Interface Firmware.
2. Toggled power off then on.
3. DAPLink started fine.
4. Enabled Automation and restarted DAPLink.
5. Tried to flash DAPLink Bootloader with no success for both [1] and [2].

On another nRF52-DK I have tried to flash Bootloader update directly from JLINK but that did not work too, so I leave it with JLINK as reference for comparison.

I have ordered a needle pin connector to reflash JLINK interfaces, so I would be able to fix if anything goes wrong. I just thought there should be a way to replace JLINK with DAPLink Bootloader + Interface without using JTAG..?

[1] https://os.mbed.com/blog/entry/DAPLink-bootloader-update/
[2] https://os.mbed.com/blog/entry/DAPLink-bootloader-updates-with-external/

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 110100004420312043574641333030313430303197969903
HIC ID: 97969903
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0246
Git SHA: 0beabef8aa4b382809d79e98321ecf6a28936812
Local Mods: 0
USB Interfaces: MSD, CDC, HID
Interface CRC: 0x50152171
Remount count: 3
```

cederom, 2018-02-21:
Provided bootloader images on [1] and [2] seems invalid. I had to build bootloader (with CRC) and that worked on bare nRF52-DK JLINK -> DAPLink update after updating Interface Firmware first.

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 110100004420312043574641313031203930303897969903
HIC ID: 97969903
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0246
Bootloader Version: 0244
Git SHA: 0beabef8aa4b382809d79e98321ecf6a28936812
Local Mods: 0
USB Interfaces: MSD, CDC, HID
Bootloader CRC: 0x0cfd8bf5
Interface CRC: 0x50152171
Remount count: 1
```

I think it would be nice to provide bootloader images within the release package @c1728p9 @sg- .

[1] https://os.mbed.com/blog/entry/DAPLink-bootloader-update/
[2] https://os.mbed.com/blog/entry/DAPLink-bootloader-updates-with-external/

jwyune, 2018-04-24:
Any update on this?

I made my own version of an interface board based on SAM3U2C. I need to install a bootloader and then a DAPLink firmware, but those images don't seem to work.

I am using an nRF51-DK for programming my interface board.

ashok-rao, 2018-04-25:
@c1728p9 , I'm facing this issue with a ublox-NINA-B1-EVK (nRF52 based) as well. Can't update J-Link OB with DAPLink (0246_sam3u2c_ublox_evk_nina_b1_0x5000.bin). Any thoughts on this?

cederom, 2018-04-25:
Is this anyhow related to https://github.com/ARMmbed/DAPLink/issues/349 and https://github.com/ARMmbed/DAPLink/issues/350 ? Are you sure @jwyune @ashok-rao  you are using stock bootloader and not anything that was updated in the meantime?

Asking @c1728p9 and @sg- to provide additional bootloaders should solve the issue..?

ashok-rao, 2018-04-25:
@cederom ..yes, I can confirm that I'm using stock bootloader.. haven't updated anything anytime. Just tried to update DAPLink to 0246, but without much success. My board doesn't even get into BOOTLOADER mode.. (holding reset, plug cable in)..

cederom, 2018-04-25:
Have you tried this? https://os.mbed.com/blog/entry/DAPLink-bootloader-update/

ashok-rao, 2018-04-26:
@cederom , looks like there will not be DAPLink support for NINA-EVK. @c1728p9 provided me info on this.

jwyune, 2018-05-08:
@cederom What stock bootloader are you talking about? I simply downloaded this file, 0244_sam3u2c_bl_0x5000.bin, and followed the direction on [2].

I am using nRF52-DK for flashing my empty SAM3U2C board, and I get this message:
```
Jiwons-MacBook-Air:~ jiwonyune$ pyocd-flashtool ./Documents/Projects/mbed/_ref/Bootloaders\ and\ firmwares/0244_sam3u2c_bl_0x5000.bin -t nrf52
INFO:root:DAP SWD MODE initialised
Traceback (most recent call last):
  File "/usr/local/bin/pyocd-flashtool", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/site-packages/pyOCD/tools/flash_tool.py", line 152, in main
    frequency=args.frequency, blocking=False)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/board/mbed_board.py", line 265, in chooseBoard
    mbed.init()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/board/board.py", line 51, in init
    self.target.init()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/target/coresight_target.py", line 74, in init
    self.dp.init()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/coresight/dap.py", line 112, in init
    self.read_id_code()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/coresight/dap.py", line 117, in read_id_code
    self.dpidr = self.read_reg(DP_REG['IDCODE'])
  File "/usr/local/lib/python2.7/site-packages/pyOCD/coresight/dap.py", line 130, in read_reg
    return self.readDP(addr, now)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/coresight/dap.py", line 203, in readDP
    return readDPCb()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/coresight/dap.py", line 194, in readDPCb
    result = result_cb()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 564, in read_reg_cb
    res = transfer.get_result()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 116, in get_result
    self.daplink.flush()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 497, in flush
    self._read_packet()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 650, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 333, in decode_data
    data = self._decode_transfer_data(data)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 279, in _decode_transfer_data
    raise DAPAccessIntf.TransferError()
pyOCD.pyDAPAccess.dap_access_api.TransferError
```

Any idea what's going on?

tcpipchip, 2018-05-14:
I am having the same problem! Shows a "drive" but not the MANUTENANCE message :(

c1728p9, 2018-05-16:
Hi @jwyune, you won't be able to flash the sam3u2c with pyOCD as pyOCD does not support that target. You'll need to use another flash programmer.

c1728p9, 2018-05-16:
@tcpipchip what board are you using and which firmware did you try and load?

tcpipchip, 2018-05-17:
Hi, i am using UBLOX EVB NINA B112

Please, send me bootloader firmware and dap software.

Btw, your DAP version is compatible with JLINK ?

Thank you!

Miguel

jwyune, 2018-05-19:
@andrewc-arm What device do I need to flash the bootloader into SAM3U?

tcpipchip, 2018-05-19:
I am using SEGGER JLINK to burn! Next week i will test the NINA burning (SWD)

tcpipchip, 2018-05-22:
Working NICE DAPLINK with EVK NINA B112
I can drag and drop and burn the NINA B112, using the DAPLINK on ATSAM3u

tcpipchip, 2018-06-01:
Hi, i think that i found a bug! I have .hex to NIBA B112, when i burn with JLINK,the NIBA B112 runs well the hex program.
But, if i use the DAPLINK, the NIBA B112  does not run this hex program.
Did somenone get this problem ?

closed 2019-01-04 by brianesquilona

cederom, 2019-06-04:
Got again into the same problem. Used `pyOCD` + `CMSIS-PACK-MANAGER` to flash Bootloader and Interface firmware using external probe (another `nRF52DK` + custom cable with that magic pogo-pins header). Bootloader is still not provided with release, it can be found at [1].

```
(venv36mac) iCeDeROM% pyocd pack -f "*sam3u2*"
  Part        Vendor   Pack       Version
-------------------------------------------
  ATSAM3U2C   Keil     SAM3_DFP   1.2.2
  ATSAM3U2E   Keil     SAM3_DFP   1.2.2

(venv36mac) iCeDeROM% pyocd pack -i ATSAM3U2C
Downloading packs (press Control-C to cancel):
    Keil::SAM3_DFP::1.2.2

(venv36mac) iCeDeROM% pyocd-flashtool -t ATSAM3U2C -a 0x80000 0244_sam3u2c_bl_0x5000.bin
WARNING:pyocd.tools.flash_tool:pyocd-flashtool is deprecated; please use the new combined pyocd tool.
INFO:pyocd.board.board:Target type is atsam3u2c
INFO:pyocd.coresight.dap:DP IDR = 0x2ba01477 (v1 rev2)
INFO:pyocd.coresight.ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
INFO:pyocd.coresight.rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=000 part=000)
INFO:pyocd.coresight.rom_table:[0]<e000e000:SCS-M3 class=14 designer=43b part=000>
INFO:pyocd.coresight.rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
INFO:pyocd.coresight.rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
INFO:pyocd.coresight.rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
INFO:pyocd.coresight.rom_table:[4]<e0040000:TPIU-M3 class=9 designer=43b part=923 devtype=11 archid=0000 devid=0:0:ca0>
INFO:pyocd.coresight.cortex_m:CPU core #0 is Cortex-M3 r2p0
INFO:pyocd.coresight.dwt:4 hardware watchpoints
INFO:pyocd.coresight.fpb:6 hardware breakpoints, 4 literal comparators
[====================] 100%
INFO:pyocd.flash.loader:Erased chip, programmed 19968 bytes (78 pages), skipped 512 bytes (2 pages) at 8.69 kB/s

(venv36mac) iCeDeROM% cp 0254_sam3u2c_mkit_dk_dongle_nrf5x_0x5000.bin /Volumes/MAINTENANCE

(venv36mac) iCeDeROM% cat /Volumes/DAPLINK/DETAILS.TXT
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 11013602442031204a353043323031323131303897969903
HIC ID: 97969903
Auto Reset: 0
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0254
Bootloader Version: 0244
Git SHA: f499eb6ec4a847a2b78831fe1acc856fd8eb2f28
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0xb7eef2ad
Interface CRC: 0x02bba193
Remount count: 0
URL: https://mbed.org/device/?code=@U?version=@V?target_id=@T
```

[1] https://os.mbed.com/blog/entry/DAPLink-bootloader-updates-with-external/