ARMmbed/DAPLink #415: Does LPC11U35 DAPLink drag and drop programming with nRF52832 connected work?

ntavish, 2018-04-02:
Does LPC11U35 supports drag and drop programming, I am testing with LPC11U35 based DAPLink connected to an nRF52832.

I have been using [mtb nina b112](https://github.com/ARMmbed/DAPLink/blob/d38c9bb8193c02b7caeafec957fe1b1f05a5dedf/records/board/mtb_nina_b112.yaml) image but Drag and drop programming doesn't seem to work. It fails with `The transfer timed out.`, with `archpro` firmware, the reason is `Could not reset target`.

cederom, 2018-04-02:
Hey @ntavish, at the moment all nRF boards use SAM3U as more powerful and resourceful HIC.. it has more Flash/RAM for DAPLink to operate.. some of its GPIO is used to identify a nRF target.. so there is a noticeable hardware configuration difference in this case. Also RESET handling is different for current nRF targets implementation..

I have created a RESET UNIFY proposition https://github.com/ARMmbed/DAPLink/pull/371 that should allow you to work even without hardware reset line attached with all HIC and TARGET.. that should solve your `could not reset target` problem.. please try that one and let us know if that works for your nRF+LPC11U35 :-)

Remember you can always use generic CMSIS-DAP HID/Debug channel to program different targets using `pyOCD` :-)

ntavish, 2018-04-02:
Hi, thanks for the response. I do not have access to Keil, is there some way to get binaries from your branch that I can test?  `mtb nina b112` is the closest to my hardware I think. At this point drag and drop is most important functionality for us, but I did test pyOCD to be working. So even if there is a non upstream build for this, that would be helpful for me, since we ordered parts for multiple boards!

cederom, 2018-04-02:
Sure, I will make a build in a free moment and share the binaries. If you need a specific build please make a PR and I will build it for you upon request. Who do you work for? :-)

cederom, 2018-04-02:
Btw if you are creating a new design please consider selecting more powerful future-proof HIC instead 11U35 which already has lots of limitations..

therealprof, 2018-04-03:
@ntavish 

> Multiple DAPlink boards have nordic chips, including DIPDAP (which mentions nRF51822) and BLE Nitrogen which says drag and drop is supported (maybe it was working at some point, when the project was called CMSIS-DAP?), so we went with LPC11U35.

LPC11U35 is not a **Nordic** chip but from **NXP**. Other options from NXP which sail under the "more future-proof" banner are the K20, K22 and the KL26Z (this one is e.g. used on the micro:bit boards).

ntavish, 2018-04-03:
> LPC11U35 is not a Nordic chip but from NXP

Yes, I meant they "support" noridc chips, the DIPDAP (based on LPC11U35) for example mentions it has prebuilt images for nRF51822, and BLE Nitrogen is LPC11U35 plus nRF52832 and mentions drag and drop support.

> the "more future-proof" banner are the K20, K22 and the KL26Z

Will keep this in mind for next iteration, thanks.

therealprof, 2018-04-03:
@ntavish Check out http://tech.microbit.org/hardware/schematic/ . They're using a Nordic nRF51822 as host MCU and KL26Z as DAPLink interface. There're also ready made images for that.

jwyune, 2018-05-08:
@ntavish I think this link might help you: https://github.com/makerdiary/nrf52832-mdk

Or this one: https://wiki.makerdiary.co/nrf52832-mdk/daplink/

NRF52832-MDK is already an mbed product available on the market, which uses LPC11U35 as an interface.

[image]

MDK provides almost everything you need for updating your DAPLink firmware on their website. I hope you find this helpful.

ntavish, 2018-05-08:
Thanks @jwyune! I'll check this out, looks like it should work.

closed 2018-12-28 by brianesquilona