ARMmbed/DAPLink #418: DAPLink without RTOS?

xcvista, 2018-04-10:
Is there any feasible way to build DAPLink without RTOS involvement? This would allow even cheaper debug interfaces, for example one based on STM32F042F6P6 (32kB Flash, TSSOP-20 package) or even ATSAMD11C14 (16kB Flash, SO-14 package) chips

If so, what files need be extracted? If not, is there any pointer or documentation on how to reimplement the CMSIS-DAP protocol?

cederom, 2018-04-10:
Hey @xcvista, not really, this is part of DAPLink design.. the RTOS overhead is really minimal and gives lots of versatility.. one day even DAPLink may be rewritten to work on ARM mbedOS..

CMSIS-DAP code is kind of submodule located at: https://github.com/ARMmbed/DAPLink/tree/master/source/daplink/cmsis-dap 

You may want to rewrite that part with HID Debug or UMS Flashing plus the USB stack if you think your time is worth the result.. still $1 IC seems cheaper..

xcvista, 2018-04-11:
@cederom I have a working USB stack already that targets extremely small STM32 chips and it can be entirely interrupt pumped (so it don't waste CPU cycles unless need to) The HID debug is what I need to run on top of that. As of the UMS I have a drastically different use for it, being used for storing the program in the box and flash targets offline, so the DAPLink UMS Flash won't work for me as-is. I also need a separate CDC channel.

xcvista, 2018-04-11:
The STM32F042 chips are what I found as "just enough peripherals" chips: it has the perfect amount of pins and peripherals to implement the features I have in mind, and stepping to any other model means introducing unused peripherals or a lot of unused pins. Too bad the F042 doesn't have a lot of Flash to accommodate the RTOS, hence I am seeking bare metal solutions.

therealprof, 2018-04-11:
@xcvista I agree. I love that chip, I basically throw it on every design I do just for good measure and it really has a killer selection of peripherals. The next step up would be the F072 which also has more flash but it's not available as TSSOP. Being able to use that as DAPLink programmer would be outstanding.

cederom, 2018-04-11:
You may also talk to Cypress Semiconductor, they have some sort of peripheral matrix in their PSoC, and you may be able to negotiate good price per chip that only uses selected peripherals :-)

therealprof, 2018-04-11:
@cederom Yeah, but they're awkward to use and they only support development with a GUI based Windows-only IDE which sucks rocks. They also have their own CMSIS-DAP implementation called KitProg but that's mostly 5V only and the chips are really somewhat tricky to use in own designs.

cederom, 2018-04-11:
Yea, STM ROX, and they have really good support for Open-Source :-) My friends that squeeze every penny out of design told me about the trick with Cypress, but I did not use that myself, just a hint for you guys :-)

flit, 2019-02-06:
Closing issue. The DAPLink interface will always use an RTOS as an integral part of its design. The RTOS will become even more important as we add more advanced features such as power monitoring and IO expansion. There is a chance that the bootloader might become RTOS-less in order to reduce size, but no guarantee.

closed 2019-02-06 by flit