ARMmbed/DAPLink #442: web usb build?

edbaafi, 2018-06-11:
Is there ay chance we could get an early webusb build now that it has been merged into master?

The cheapest Keil license that seems it would be able to build this is $1k+ for 1 year which is pretty prohibitive just to use for this project.

Thanks,
Ed

AlphaArctus, 2018-07-02:
Likewise. It would be great to get access to a new release including webusb. 

In our case we'd like to get a start on setting up and testing some in-browser editors for the Micro:Bit's to support our local primary schools but don't have access to the Keil license that we'd need to make this one off build.

Otherwise any ETA on when the next release might be would be appreciated :)

Cheers

edbaafi, 2018-07-02:
Hey Alpha,

We built this a few days ago: https://github.com/modkit/DAPLink/releases/tag/webusb  IIRC you want to use the `kl26z_microbit_if_crc.hex` file.

Seems to work with this example https://armmbed.github.io/dapjs/examples/web.html on Mac and Chrome OS but not on Windows.  Didn't dig into that yet.

Best,
Ed

AlphaArctus, 2018-07-02:
Awesome! Thanks Ed.

Just tested it out under Linux+Chrome. Works a treat :+1: 

Much appreciated :)

jaustin, 2018-08-06:
The latest build, 0247, contains webUSB for the micro:bit and is built from the ARMmbed source.

https://github.com/ARMmbed/DAPLink/releases/tag/v0247

edbaafi, 2018-08-08:
Thanks Jonathan,

We should probably close this issue out.  But I have one question.  What do you mean by `built from the ARMmbed source`.  Does this mean one can use any mbed supported compiler like ARM_GCC and Keil is no longer needed?  If so can we get an update to the dev guide so others can build in the future? https://github.com/ARMmbed/DAPLink/blob/master/docs/DEVELOPERS-GUIDE.md

Thanks,
Ed

closed 2018-12-28 by brianesquilona