ARMmbed/DAPLink #443: Error in Greentea

jwyune, 2018-06-24:
I made a few custom boards, based on Maxim Integrated's chips and Nordic's chips. I am trying to run Greentea tests (https://github.com/ARMmbed/greentea) for validation, but it seems like all I'm getting are just errors.

[SDT32620_log.txt](https://github.com/ARMmbed/DAPLink/files/2131322/SDT32620_log.txt)
[SDT32625_log.txt](https://github.com/ARMmbed/DAPLink/files/2131323/SDT32625_log.txt)
[SDT51822_log.txt](https://github.com/ARMmbed/DAPLink/files/2131324/SDT51822_log.txt)
[SDT52832_log.txt](https://github.com/ARMmbed/DAPLink/files/2131325/SDT52832_log.txt)

You can see error logs in the attached files.

**I wonder if DAPLink firmwares are critical for Greentea tests.** I didn't make DAPLink firmwares myself; I am using DAPLink firmwares provided by the community, not specifically for my custom boards. 

For example, I am using MAX32620PICO for my Maxim Integrated's chips. I am using LPC11U35 for nRF51822 and nRF52832. Hence, when I do `mbed detect`, it doesn't recognize my target. However, I have no problem running other codes on my custom boards; it is just Greentea tests that are causing this problem.

Please let me know if you have any clue.

Thanks a lot in advance.

cederom, 2018-06-25:
Hey @jwyune, greentea [1] is a mbed-os testing framework.. you problems seem to be related with build issues itself (linker issues specifically) not really with DAPLink.. have you tried asking this question on project page [1]?

Some MbedOS tools depends on `mbed-ls` which detects connected boards. When boards cannot be detected or are detected incorrectly (i.e. no given ID or invalid ID) some problems may occur..

[1] https://github.com/ARMmbed/greentea

andrewc-arm, 2018-06-25:
Hi, @jwyune and @cederom 
Now I review your comment and I think this problem is related with board ID/name pairing issue. 
It seems that new DAPLink interface + SDT target needs new board ID/name.
I will discuss internally and come back to this issue.

jwyune, 2018-06-26:
Thanks to both of you, @cederom and @andrewc-arm 

I look forward to your reply.

andrewc-arm, 2018-07-02:
Hi,

I just enabled SDT's module name and IDs.

SDT’s Board Name | Internal MCU | ID
-- | -- | --
SDT32620B | Maxim MAX32620 | 3101
SDT32625B | Maxim MAX32625 | 3102
SDT51822B | Nordic nRF51822 | 3103
SDT52832B | Nordic nRF51832 | 3104
SDT64B | NXP K64F | 3105
SDT8195B | Realtek RTL8195AM | 3106
SDT32427B | STM STM32F427 | 3107
SDT32429B | STM STM32F429 | 3108
SDT32437B | STM STM32F437 | 3109
SDT32439B | STM STM32F439 | 3110

However, I need to consult internally on how to inject those IDs to DAPLink FW or boards so that 'mbed detect' will tell us it's SDT's modules. I will feedback to you, soon.

Thanks,

cederom, 2018-07-02:
Hey @andrewc-arm, but you do confirm the problem is related to missing ID's of the new/unknown boards?

andrewc-arm, 2018-07-03:
Hi, @cederom 
I suspect that this issue is related with not-yet-ready ID/name match. Would you want to close this ticket as it is or would you want to wait for the DAPLink ID injection process to see if it really resolves?

Hi, @jwyune 
In the separate email I am going to send you, I will guide you on injecting the new IDs to DAPLink firmware and changing mbed-ls with a github pull request.

Thanks,

cederom, 2018-07-03:
I guess confirming this problem and working solution would be desirable for other folks that get into this problem in futute too..? :-)

closed 2019-01-04 by brianesquilona