ARMmbed/DAPLink #444: Is there c library which can communicate with DAPLink just link pyOCD ?

trlsmax, 2018-06-25:
Hi, 
Is there c library which can communicate with DAPLink just link pyOCD ?

trlsmax, 2018-06-25:
I just implement a simple SEGGER RTT viewer for stlink, so I think if there is a library I can use to talk to DAPLink and read / write the memory of the target MCU , I can add DAPLink support.

The code is base on C, so pyOCD is not an option.

cederom, 2018-06-25:
pyOCD does lots of low level dap/debug work for you.. but you can manually use the HID channel to perform operations on your own if you like according to CMSIS-DAP protocol standard.. if you work with ST-LINK already you probably already use some sort of HID/USB library.. if they use CMSIS-DAP porting code to work with DAPLink should be relatively simple.. good luck! :-)

closed 2019-01-04 by brianesquilona