ARMmbed/DAPLink #446: Adding a new HIC,Which script files need to be modified?

neomissing, 2018-06-26:
I have adapted a new mcu, the program is right, but there will be problems with script compilation, can not generate bin and hex files, I would like to know which script files need to be modified to support the new mcu compilation？

cederom, 2018-06-26:
Hey @neomissing, the only current documentation is here: 
https://github.com/ARMmbed/DAPLink/tree/master/docs
You may want to see "Adding New Board" and "Adding New Target" in "Porting" documents :-)

neomissing, 2018-06-27:
Adding a new HIC,Which script files need to be modified?

cederom, 2019-01-04:
Maybe you can find some of my docs useful..

Here is sliced old-way docs:
https://github.com/cederom/DAPLink/tree/cederom_doc/doc

Here is all in one version:
https://github.com/cederom/DAPLink/blob/cederom_doc/doc/DAPLinkHandbook.md