ARMmbed/DAPLink #447: How to change bootloader and interface version string?

YutingYou, 2018-06-27:
I changed DAPLINK_VERSION=1 in both bootloader and interface, but still got old version.

```
# DAPLink Firmware
Unique ID: 0000000033508829023032393931394ea5a5a5a597969905
HIC ID: 97969905
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Daplink Mode: Bootloader
Bootloader Version: 0001
Interface Version: 0242
Local Mods: 1
USB Interfaces: MSD
Bootloader CRC: 0x04cbc692
Interface CRC: 0xede36280
Remount count: 0
```

flit, 2018-06-27:
Looks like the bootloader has the correct version of 0001. Are you sure you rebuilt the interface and updated it on the device?

Btw, can I ask what your use case is for changing the version? Ideally, we'd like to keep from having multiple distributions of DAPLink with different version sequences. However, we recognize that there are some cases where declaring derivative and/or target-specific versions would be useful. We are discussing whether and how to improve the version numbering scheme.

YutingYou, 2018-06-28:
@flit Thanks for your reply, I rebuilt both interface and bootloader. I found `DAPLINK_VERSION` is only used in details.txt, should I change some python script file which are used to create hex file?

We fork this repository and bring changes, so we need a version number to distinguish each change.If we modify the version number to be different from the version number we actually see, this can lead to misunderstandings.