ARMmbed/DAPLink #449: Support external flash

Cheong2K, 2018-06-30:
I would like to discuss if the DAPLink interface is possible to flash data to external flash, for example, the nRF52840 DK has an external QSPI flash which supports XiP. Writing to the internal flash of the nRF52840 is fully functional and their tool `nrfjprog` is possible to flash data (.hex) to the external flash and it maps the external flash to address 0x12000000. The ideal use case is, if I drag the hex with data at address 0x12000000 then it will flash it to the external flash, then it will be perfect!

cederom, 2018-06-30:
Hello @Cheong2K! That should be possible, FlashAlgo is a binary blob uploaded into RAM and executed from there and it writes chunks of data transmitted from USB to RAM into Flash memory. It should be possible to modify FlashAlgo so it also writes to SPI for this particular target (family)..

brianesquilona, 2019-01-04:
@cederom Thanks for the comment.

closed 2019-01-04 by brianesquilona