ARMmbed/DAPLink #454: GR_LYCHEE - programming fails with "Flash algorithm write command FAILURE"

studavekar, 2018-07-12:
GR_LYCHEE fails to programme firmware with the latest interface update 0246 - PR https://github.com/ARMmbed/DAPLink/pull/451 

Contains of FAIL.txt
```
error: Flash algorithm write command FAILURE
type: target 
````

Daplink details : 
```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 5501000002e3e62500000000000000000000000097969902
HIC ID: 97969902
Auto Reset: 0
Automation allowed: 1
Overflow detection: 1
Daplink Mode: Interface
Interface Version: 0246
Git SHA: af0dbcb73fa01ad7ab39a14d9648a41d83dca51f
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Interface CRC: 0x529ccf5e
Remount count
```

cc : @maclobdell @toyowata @d-kato

d-kato, 2018-07-13:
It seems to be the effect of this change.
https://github.com/ARMmbed/DAPLink/pull/451/commits/9ab32b06ef945abca633074747b999ab86f4ade4

It is programmatically equivalent, but the link order may be related. I will check it a bit more.

d-kato, 2018-07-13:
It is because the file name was changed to "validation.c".
There is another file with the same name, and functions "validate_bin_nvic" and "validate_hexfile" were not successfully overridden.
I change the file name from "validation.c" to "validation_rza.c".

flit, 2018-07-27:
Closing issue after merging fix in #455.

closed 2018-07-27 by flit

fabioriksilva, 2020-04-30:
I'm trying to make the:
     # DAPLink Firmware - see https://mbed.com/daplink
**Unique ID: 123400011be6e73900000000000000000000000097969902
HIC ID: 97969902
Auto Reset: 1
Automation allowed: 1
Overflow detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0254
Git SHA: d15cc3d2b39c550842f658593e4b7980f6e06148
Local Mods: 1
USB Interfaces: MSD, CDC, HID, WebUSB
Interface CRC: 0xe169a5a0
Remount count: 2
URL: https://mbed.org/device/?code=@U?version=@V?target_id=@T**

   Work with the target LPC1768 and interface LPC11U35, but, it has taken more than 3 months and the problem still is going on: 
   It doesn't matter what I try, it replies always:

  **error: Flash algorithm write verify command FAILURE
  type: target**

   I have tried as above in this topic, but no success.
   The code is being compiled by keil REV5_10. The environment is perfect, at least I think it, because when the interface is used as CMSIS it works pretty well.

  Could someone help-me with this problem?

  Thanks a lot!