ARMmbed/DAPLink #470: micro:bit LED behaviour change between 0242 and 0243

jaustin, 2018-08-06:
_(We didn't pick this up until now because only small numbers of quite technically-minded people have been upgrading their micro:bit firmware, however, with the addition of webUSB we're expecting a much greater number of people to try the upgrade)_

Until 0243 the micro:bit's only LED was always on when the micro:bit was powered over USB.

After 0243 the LED blinks 'normally' at initialisation and then turns off.

0241 is currently shipping on micro:bits, so for all existing devices the behaviour is that the LED stays on.

A quick look at the commits between those two releases (https://github.com/ARMmbed/DAPLink/compare/eab079b8a1326f0a3647e3df33cf5dda13994a1f...b403a07) led me to take a suspicious look at:
https://github.com/ARMmbed/DAPLink/commit/e9422708f972dca84c4475ca3ac48a9e5e49af72#diff-36cc30e57349a6d5cca2b27916ae65a8
and
https://github.com/ARMmbed/DAPLink/commit/3259920eeb326c3aabbfe2ec274bad88c5dc767f

micro:bit only has one LED, and we worry that people will be confused by the change (in fact, this was raised by people testing 0247 for webUSB). As we want to change the default firmware version shipped on the micro:bit to a recent release, the change in behaviour is an issue.

flit, 2018-08-06:
Does the micro:bit LED have to blink on USB activity?

flit, 2018-08-06:
What is the priority of this issue? How quickly do you need a release that includes it?

flit, 2018-08-07:
PR #473 fixes this problem. The micro:bit LED state is set to on. USB activity will still blink the LED, then it will return to on.

closed 2018-08-07 by flit

jaustin, 2018-08-08:
Awesome, cheers! Sorry I missed the priority question - I don't think this is something to run a new release for right away as we have the issue discussed but as-yet-not-filed about name of micro:bit in webUSB pairing dialogue too. I'll file that later on today.