ARMmbed/DAPLink #478: Boards are not well identified in webUSB dialogue

jaustin, 2018-08-15:
_This is potentially a greater issue for micro:bit than it is for other platforms, but as webUSB becomes more prevalent I think smooth user experience will help._

When you use webUSB the browser shows you a 'pairing' dialogue that includes the name of your device:

[image]

Many users of micro:bit have no idea what DAPLink and CMSIS-DAP are, so if that's the only information there then it will confuse them. We're keen to leave educational breadcrumbs, so keeping those strings there additionally is fine, but it would really help for the dialogue to say 'BBC micro:bit' somewhere

Part A of this is:
Is it possible for the micro:bit to show up as a BBC micro:bit in this dialogue please?

Part B is a bit more complex:
If someone has two devices plugged in at once, how will they know which one to program? Micro:bit has a built in 'Bluetooth paring mode' where we map a bar graph on the screen to a text string, using the following mapping 

[image]

The name is calculated based on the nRF's serial number as show here
https://github.com/lancaster-university/microbit-dal/blob/5746eb94a8fbe944f474c89c3edd237731cb0f0a/source/core/MicroBitDevice.cpp#L94

One possible way to help people identify multiple micro:bits would be to name each one using the method above, by computing the 'friendly name' in DAPLink (I believe we already have the serial number in DAPLink and there are a bunch of other reasons the friendly name would be useful, for example in details.txt).

If we did this then 'BBC micro:bit [zuvap]' would be a great description (and could include 'CMSIS-DAP' or 'DAPLink' too if necessary.

@arekzaluski, @microbit-sam and @microbit-carlos

jaustin, 2018-08-28:
@brianesquilona in #479 you mentioned that you are working on something like this - any thoughts on whether it'll include the micro:bit 'friendly name'?

brianesquilona, 2018-08-28:
#488 

@jaustin , it will not be straight forward to support this naming on runtime, so currently only compile time is supported.

closed 2018-12-28 by brianesquilona