ARMmbed/DAPLink #483: V - Micro:bit

bsimmo, 2018-08-18:


carlosperate, 2018-08-20:
Is the micro:bit display scrolling "pipev"? In that case that would be from the last program flashed to the micro:bit, DAPLink doesn't have access to the LED matrix.
When flashing a DAPLink hex file into maintenance mode, does the flashing complete successfully? Is there fail.txt in the drive after the micro:bit disconnects and connects again?

bsimmo, 2018-08-20:
Yes on the display.

I'll have a look again when I can (not near it at the moment).
I do have a few others to test on.

I dropped it over, waited a while, I think it went ping pong in Windows and
then I disconnected and reconnected

I'll wipe out any old drivers/attachments hidden in Device Mange as wellin
case they are stopping it from seeing the device.

End of the week hopefully.

bsimmo, 2018-10-09:
For the info, I tried to fix it back then but nothing worked.
Left it for a while (akak until today)

Downloaded the new firmware, place a test one in Maintenance and flashed (v0250).
Worked, and tested ith simple MakeCode hex.
So flashed the pipeV one and it flashed fine and worked etc..
case closed.

Of course no idea if a windows10 update fix along the way or a firmware fix.

closed 2018-10-09 by bsimmo