ARMmbed/DAPLink #484: develop env initialize fail

wuxx, 2018-08-26:
I follow the instruction in DEVELOPERS-GUIDE.md to init the compile, but when I call the 
progen generate -t uvision, it return fail, here is the output，can somebody help me figure out what's wrong here ?
(venv) E:\oss\ARMmbed\DAPLink>progen generate -t uvision
Traceback (most recent call last):
  File "c:\users\44783\appdata\local\programs\python\python36\Lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\44783\appdata\local\programs\python\python36\Lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "E:\oss\ARMmbed\DAPLink\venv\Scripts\progen.exe\__main__.py", line 9, in <module>
  File "e:\oss\armmbed\daplink\venv\lib\site-packages\project_generator\main.py", line 62, in main
    return args.func(args)
  File "e:\oss\armmbed\daplink\venv\lib\site-packages\project_generator\commands\generate.py", line 30, in run
    if project.generate(args.tool, copied=args.copy, copy=args.copy) == -1:
  File "e:\oss\armmbed\daplink\venv\lib\site-packages\project_generator\project.py", line 552, in generate
    files = exporter(self.project['export'], self.settings).export_project()
  File "e:\oss\armmbed\daplink\venv\lib\site-packages\project_generator\tools\uvision.py", line 487, in export_project
    path, files = self._export_single_project('uvision') #todo: uvision will switch to uv4
  File "e:\oss\armmbed\daplink\venv\lib\site-packages\project_generator\tools\uvision.py", line 466, in _export_single_project
    uvoptx_dic = xmltodict.parse(open(self.uvoptx_file))
  File "e:\oss\armmbed\daplink\venv\lib\site-packages\xmltodict.py", line 328, in parse
    parser.ParseFile(xml_input)
TypeError: read() did not return a bytes object (type=str)

(venv) E:\oss\ARMmbed\DAPLink>

wuxx, 2018-08-26:
after google, I modify the  uvision.py in DAPLink\venv\Lib\site-packages\project_generator\tools\
line 466 
from
        uvoptx_dic = xmltodict.parse(open(self.uvoptx_file))
to
        uvoptx_dic = xmltodict.parse(open(self.uvoptx_file, "rb"))
 this command didn't report fail, it generate the projectfiles\uvision\ directory, but the uvproj it generate seems still cannot work

wuxx, 2018-08-26:
PS: the python version on my host is 3.6.2

cederom, 2018-08-27:
Have you tried Python 2?

0xc0170, 2018-08-27:
@wuxx python 2 should work , I'll check the python 3 (earlier version were working previously)

0xc0170, 2018-08-27:
@wuxx I fixed it upstream (https://github.com/project-generator/project_generator/pull/442), will release a new patch addressing this mode issue. tests are green again for python 3.x. version 0.9.12 should address the issue with python 3.4 and above

Not certain what issue you are facing now. I locally used python 2.7 and 3.7. The projects are building and looks correct.

wuxx, 2018-08-28:
@0xc0170 here I found the modification https://pythonadventures.wordpress.com/tag/xmltodict/  , there is no much reason why it told. 
and I will try the new patch your release, thank you

wuxx, 2018-08-28:
and here is one project it generate.
[lpc11u35_archlink_if.zip](https://github.com/ARMmbed/DAPLink/files/2326477/lpc11u35_archlink_if.zip)
[image]
I use uVision V5.15, and when i click the compile, here is the output, actually compile didn't start.

wuxx, 2018-08-28:
and I notice another problem, there is a pre_build_script.bat will run before the compile, but this py script run fail, because it is a python 2 style script, for example
print "#> Pre-build script start"
should add the brackets, to be
print ("#> Pre-build script start")
then it can run under python 3.x

did you notice this problem, or this is because the project I generate is already wrong ?

0xc0170, 2018-08-28:
What's your progen version you are testing?

Here's my setup:
- progen 0.9.12 (that fixes python 3.x)
- uvision 5.17
- python 2.7 or python 3.7 (testing both)

here's the output:

[image]

Compilation working. See there the first step, runs a python script to get some git info. I suspect you are having problems with git (how did you clone the repo ?)

wuxx, 2018-08-28:
I use git on windows, here is the snapshot
[image]

wuxx, 2018-08-28:
Hi @0xc0170
and I try the develop init on another win7 machine (python 3.7.0, git 2.14.1, DAPLink master branch)
when I call the 
virtualenv venv
there is another fault. :(
[image]

wuxx, 2018-08-28:
Hi @0xc0170 I try another new PC (yes, a third PC, win10 OS), install the python 2.7.15, git version 2.18.0
and this time, it works!
I notice the previous git project I clone is 
https://github.com/ARMmbed/DAPLink.git
but in the  developers guide the git project is 
https://github.com/mbedmicro/DAPLink
this time I clone the  latter。
the other step is same with before。
this two project is not same ？
and I cannot search the latter project on github。

0xc0170, 2018-08-28:
> I notice the previous git project I clone is
https://github.com/ARMmbed/DAPLink.git
but in the developers guide the git project is
https://github.com/mbedmicro/DAPLink
this time I clone the latter。

They point to the same (both repositories). 

Seems like env problem to me (the screenshot above).

I think we resolved dependency python 3 issue.

brianesquilona, 2018-12-28:
Closing this issue as it can't be reproduce in current environment.

closed 2018-12-28 by brianesquilona