ARMmbed/DAPLink #485: Android identifies DAPLink as a corrupted USB drive

microbit-carlos, 2018-08-28:
When a micro:bit is connected to an Android device via USB, the operating system shows a notification about the drive being corrupted, and when you tap on the notification it suggest to format it: 
 [image] [image]

This is related to issue https://github.com/ARMmbed/DAPLink/issues/269 - **Using daplink with Android over USB**,  and a fix is provided in PR https://github.com/ARMmbed/DAPLink/pull/401 - **VFS/VBR/FAT Boot Sector Signature + Bootstrap Code / Android USB OTG FIX** (thank you 
@ cederom!). However, this fix will make the micro:bit bootable and that is not a desirable feature.

The micro:bit can now be used on Android via WebUSB, which is fantastic news! But this notification about a corrupted Mbed drive can be quite scary and confusing for users. Would it be possible to somehow remove this detected corruption, but not make the drive bootable?
It would fine if DAPLink is not recognised as a MSD in Android, as long as WebUSB is still available.

cederom, 2018-08-28:
I will check if there are any structures in VBR that may disable the disk boot. There was one bit that might come handy but I need to verify that. There is no MBR on the drive so we cannot simply mark partition as inactive (that disk is one big partition with VBR instead of MBR). From what I remember `fsck-vfat` implementation on Android will assume disk is corrupted then we don't have boot signature on the VBR and that was used to mark partition non-bootable thus either-or situation. Unfortunately this partition/disk signature is used to detect disk and filesystem and its there from 80s..

flit, 2019-02-06:
There's no way we could identify the host as being Android and disable MSD.

#401 will probably be merged in soon because without it the filesystem *is* corrupt. The fact that the disk is thought to be bootable is a problem with the host machine, not DAPLink.

microbit-carlos, 2019-06-06:
Fixed by #401.

closed 2019-06-06 by microbit-carlos