ARMmbed/DAPLink #486: DAPLink WebUSB identified as LPC1768 on first connection

microbit-carlos, 2018-08-28:
Opening this issue to separate the side-discussion in https://github.com/ARMmbed/DAPLink/issues/479.

When a micro:bit with DAPLink with WebUSB is plugged for the first time, chrome will list the name of the WebUSB device as `LPC1768 `:

[image]

And once it's paired it will be listed in Chrome as a USB device without a name:

[image]

To replicate follow the steps below, assuming you are on a Windows 10 computer that has successfully used a WebUSB enabled micro:bit already:
* Disconnect any DAPlink/Mbed device from the computer
* Go to https://makecode.microbit.org/beta?webusb=1#editor
* Revoke access to all previously connected devices to this website
    - At the left of the URL click on "secure"
    - Click on the cross for each entry on the list:
    [image]
* Plug in a micro:bit
* Go to the Device Manager and remove all DAPLink drivers
    - On the `Ports (COM & LPT)` list find the `USB Serial Device (COMXX)` entry, right click and `Uninstall device`
    - On the `Universial Serial Bus Controllers`, find the `USB Mass Storage Device` entry with VID 0x0d28 and PID 0x0204, right click and `Uninstall device`
    - On the `Universial Serial Bus Controllers`, find the `USB Composite Device` entry with VID 0x0d28 and PID 0x0204, right click and `Uninstall device`
* Unplug the micro:bit
* Close and reopen chrome to go again to https://makecode.microbit.org/beta?webusb=1#editor
* In MakeCode, click on the cog at the top right corned and Pair device:
    [image]
* You should see the connection window with the wrong name:
    [image]

Once you've connected once, the next time the right device name appears:

[image]

But the list of USB devices has increased instead of replacing the previous entry:

[image]

flit, 2018-09-19:
As best we can tell, this issue is caused by the DAPLink USB VID/PID being assigned the name "LPC1768" in a table of VID/PIDs that comes with Linux, and this table seems to have been reused for Chrome.

See [the current usb.ids file](http://www.linux-usb.org/usb.ids) and [live USB IDs database](https://usb-ids.gowdy.us) for the table. The current version has the name "ARM mbed" assigned to VID=0x0d28 ("NXP", should be "Arm"!!) PID=0x0204. An older version or a variant probably used "LPC1768".

My guess is that prior to pairing, Chrome simply looks up the VID/PID in this table and reports whatever name if finds. Not much we can do about that.

jaustin, 2018-10-01:
Can anyone reproduce this on a system that has *never* seen the mbed serial driver? I just tried Carlos's steps on my Win10 machine and still saw the correct name

microbit-carlos, 2018-10-01:
Just tried this in a brand new Windows VM and it did show the micro:bit name directly on first connect:
[image]
So you are onto something there.

microbit-carlos, 2018-10-01:
Okay, so even more weirdly, right after my previous message I closed Chrome and reopen it, and the second time it showed me the LPC1768 name:
[image]

mathias-arm, 2021-06-08:
@microbit-carlos: does that issue still exist?