ARMmbed/DAPLink #487: Mbed driver causes issues with WebUSB driver

microbit-carlos, 2018-08-28:
When a computer running Windows 10 (and possibly Windows 8) already has the Mbed driver installed from https://os.mbed.com/handbook/DAPLink (http://os.mbed.com/media/downloads/drivers/mbedWinSerial_16466.exe), then Windows will fail to install the WebUSB driver.

[image]

[image]

[image]

We can uninstall the mbed driver by finding all thee DAPLink entries in the Device Manager, "right-click/Uninstall device" and mark the "Delete the driver software for this device". At that point unplugging the device and replugging it will install the all the default drivers as expected.

Older versions of DAPLink needed this driver for serial to work, and a significant number of the deployed micro:bits are likely to still have v0234, we expect a portion of the existing micro:bit user base to have the Mbed drivers installed on their computers and perhaps not even realise it.

Is there anything that could be done in the the Mbed drivers or DAPLink that could avoid or fix this incompatibility?

cederom, 2018-08-28:
Is new bundle of drivers with WebUSB included that would contain all previous necessary drivers and avoid existing conflicts?

jaustin, 2018-08-28:
For most people versions since 0241 don't require drivers at all, so we'd prefer to avoid any drivers, rather than a fixed-up bundle

cederom, 2018-08-29:
How about Serial Port Driver?

microbit-carlos, 2018-08-29:
With 0241 the serial driver gets automatically installed in Windows 8/10, so there is no need to use the .exe installer.
It's possible older versions of Windows (like Win7) still require the manual installation, but WebUSB is not supported in those, so it wouldn't be a problem.

jaustin, 2018-09-03:
Can anyone reproduce this on a system that has *never* seen the mbed serial driver? I just tried Carlos's steps on my Win10 machine and still saw the correct name

abchatra, 2018-10-01:
How do I uninstall the embed driver? Is there any step by step instructions?  What is the other workaround here?

microbit-carlos, 2018-10-01:
We have this micro:bit support article with instructions on how to uninstall it: https://support.microbit.org/en/support/solutions/articles/19000089574-beta-testing-mbed-driver-doesn-t-work-with-windows

abchatra, 2018-10-01:
I have no _embed Composite Device entry_ in my device manager.  I am getting access denied error when I select the microbit from the chrome authorization dialog.

jaustin, 2018-10-02:
@abchatra do you know for sure that you've installed the mbed driver on that machine? And if so, can you share what you do have in device manager? I can't think why your setup would be different to our test one where Carlos made that help doc.

abchatra, 2018-10-02:
My issue is common in both python and makecode. When I can select a device but it gives me access denied error. 
[image]

carlosperate, 2018-10-02:
Device driver looks fine in the screenshot, might be a different issue more related to DAPJs than DAPLink.

jaustin, 2018-10-23:
@abchatra did you get any more information on this from your system? Or is it still an issue for you?

abchatra, 2018-10-23:
I haven't got cycles to get into this. I still see "Access denied". I will spend some cycles next week

0xc0170, 2019-11-29:
> We have this micro:bit support article with instructions on how to uninstall it: https://support.microbit.org/en/support/solutions/articles/19000089574-beta-testing-mbed-driver-doesn-t-work-with-windows

Had this problem as well. This is unfortunate, we should state do not use the driver for win10 in any case .

JiriAtHoneywell, 2020-06-22:
The right driver is **winusb.inf, winusb.sys**.

Search your Windows directory for the file **winusb.inf** and then install it manually.

Right click on the WebUSB: CMSIS-DAP in the Device Manager >> Update Driver >> Browse my Computer... >> Let Me Pick ... >> Universal Serial Bus Device >>Have a Disk >> Path to the winusb.inf. >> Select WinUSB Device

[image]

Kishlay-notabot, 2022-04-10:
> My issue is common in both python and makecode. When I can select a device but it gives me access denied error. [image]

same here

microbit-carlos, 2022-04-11:
The access denied error is often caused by having another MakeCode or Python editor window or tab that is already WebUSB connected (MakeCode can connect automatically, so you don't even need to click connect in some cases).

SekiBetu, 2025-03-21:
> We have this micro:bit support article with instructions on how to uninstall it: [support.microbit.org/en/support/solutions/articles/19000089574-beta-testing-mbed-driver-doesn-t-work-with-windows](https://support.microbit.org/en/support/solutions/articles/19000089574-beta-testing-mbed-driver-doesn-t-work-with-windows)

thx, it works