ARMmbed/DAPLink #492: MSD flashing issues with Edge (The transfer timed out)

microbit-carlos, 2018-08-31:
Originally reported in https://github.com/Microsoft/pxt-microbit/issues/1158, but as @DaveAtKitronik has correctly identified, this problem occurs with DAPLink 0247, but not with 0241.
I was also able to replicate both, the issue with 0247, and not having the problem with 0241.

It looks like with 0247 if we use MSC flashing in the micro:bit using the Edge "save as" option (common when downloading micro:bit programs from MakeCode or Python online editor), then quite often DAPLink times-out and creates the following `fail.txt` file on reboot:

```
error: The transfer timed out.
type: transient, user
```

To test I've just served a few hex files with python (`python -m SimpleHTTPServer`) and downloaded one of them to the micro:bit with "right-click/save as":

[image]

The file transfer finishes when the "finished downloading" message appears at the bottom of the Edge window. Normally at that point the micro:bit quickly reboots, denoted by the USB drive disappearing and appearing again in "My PC".

In this case you can see how the drive stays on much longer than usual (and the hex file still appears inside the drive) and, when it finally reboots, it contains the `fail.txt` file.

Please note: The fact that the drive rebooted after I dismissed the Edge message is mere coincidence. When the issue occurs it takes more or less the same amount of time for the reboot, no matter how long that message is displayed in Edge.

microbit-carlos, 2018-08-31:
It's quite possible that the hex file is correctly flashed, as the program does start to run before the USB drive disappears/reappears (so I assume DAPLink finishes flashing and resets the nrf?).
However, I haven't gone through the process to read back the flash to compare against the original hex file, so this assessment is based solely on the seeing the micro:bit running a different program.

DaveAtKitronik, 2018-08-31:
Pretty sure that on at least one occasion the hex did not flash correctly in my original discovery-which is what prompted me to look at the drive. Cannot try to confirm until next week. The hex I originally saw this with uses a lot of trig and floating point maths, and is quite large- not sure if that would make a difference, but might make a misflash easier to spot.

brianesquilona, 2018-09-10:
#497

microbit-carlos, 2018-09-13:
Thank you @brianesquilona for being able to push this fix to the 0249 release!

closed 2018-09-13 by microbit-carlos