ARMmbed/DAPLink #494: LPC11U35 Daplink failed to debug Cortex M7 based NXP MIMX1050 MCU

cplmaomao, 2018-09-04:
Tried to debug IMIX1050 board, but found LPC11U35 Daplink failed to connect to Cortex M7 based NXP MIMX1050 MCU?

Daplink can not read the CM7 Core ID.

I tried the latest released bin file for different interface MCU in Daplink release package.

It should be by default that daplink (no matter it build for any MCU), to support CMSIS-DAP ? That is , a Cortex-M MCU, like MIMX1050?

flit, 2018-09-05:
Hi @cplmaomao, what debug probe setup are you using, and what host software?

The MIMXRT1050-EVK/B board comes with a K20DX128-based on-board DAPLink debug probe. So I don't see where the LPC11U35 fits in the picture.

DAPLink does not itself read the CPU Core ID. The host tool such as pyOCD is responsible for reading the Core ID; DAPLink and the CMSIS-DAP protocol is just a pipe for SWD transactions. The current version 0.11.2 of pyOCD does not support Cortex-M7, but this is being worked on. Specifically, support for the MXIMRT1050 is a priority.

cplmaomao, 2018-09-06:
hi, flit,

Thanks for your reply.  

I know K20 is on MIMXRT1050 EVK board, so the released DAPLINK firmware for K20 can support all debug & drag-drop & usb cdc features.  But, i have an LPC11U35 debugger which is not on a EVK board, it is just a DIY debugger.  When i use it for debugging stm32 or kinetis , every thing goes well.  But,  when I used LPC11U35 debugger (flashed it with daplink latest firmware) to connect RT1050 board (which also is a customized board), it can not read CORE ID. I am using MDK V5.25.

flit, 2019-02-06:
Closing issue. I think the problem is either with the hardware somehow, most likely incompatible voltages. Please reopen the issue if you have more information that shows DAPLink as the suspect.

closed 2019-02-06 by flit