ARMmbed/DAPLink #496: Feasibility of daplink hosting networked "debug server"

loverdeg-ep, 2018-09-06:
I want to debug and monitor board running mbedOS remotely over a local IP based network.

Any comments on the feasibility of adding gdbserver protocol support or a "pyOCDserver" interface to daplink?

I'm essentially envisioning using a modified FRDM-K64 or an evolved version of the CI board to act as a network debug server and test runner.

flit, 2018-09-06:
Hi @loverdeg-ep. The easiest way to do this, and the only way right now, is connect the board to a PC, run pyocd-gdbserver on that PC with the `--persist` and `--allow-remote` options, then connect to it from your development machine. You could even use a Raspberry Pi as the "PC" server running pyOCD.

Serving the CMSIS-DAP protocol over a TCP port would be an interesting addition to DAPLink, but won't happen in the short term. I have added it to the todo list, though! 😀

loverdeg-ep, 2018-09-06:
@flit

Yes, currently using an in-house SBC versus PI.

Glad to see it has been added to the list.

This would be a particularly nice feature when used with an Eclipse Che workspace, custom POE CI board, and CI system.

cz172638, 2021-05-14:
there is [esp8266-cmsis-dap implementation which uses usbip](https://github.com/windowsair/wireless-esp8266-dap) which can be used for inspiration