ARMmbed/DAPLink #506: Non-continuous address

microbit-carlos, 2018-09-26:
When flashing via drag and drop of a hex file, DAPLink needs the addresses to be sequential: 

https://github.com/ARMmbed/DAPLink/blob/ab7abed209eb166a98a8ecc69b05a4dcbe86adc6/source/daplink/drag-n-drop/flash_manager.c#L127-L134

If we have a file that first tries to write to address `0x1000_10C0` and then to `0x0000_0000` (example file [uicr-micropython-pycode.hex.txt](https://github.com/ARMmbed/DAPLink/files/2420961/uicr-micropython-pycode.hex.txt)) we get an clear error in the form of a fail.txt and assert.txt files:

```
error: An internal error has occurred
type: internal
```

```
Assert
File: ..\..\..\source\daplink\drag-n-drop\flash_manager.c
Line: 131
Source: Application
```

This is fine, but if we first start to write from address `0x0000_0000`, then jump to address `0x1000_10C0` and then jump back to `0x0003_E000` (example file [micropython-uicr-pycode.hex.txt](https://github.com/ARMmbed/DAPLink/files/2420966/micropython-uicr-pycode.hex.txt)), then the flashing fails silently.

The block starting at `0x0000_0000` and `0x1000_10C0` are flashed correctly, but the block starting at `0x0003_E000` is not flashed, which is expected, but there is no fail.txt or asset.txt files present.

Tested with DAPLink 0243 and 0250, in a micro:bit.

A test on DAPLink 0241 produces an assert.txt file but no fail.txt.

Although I understand why the sequential addresses enforcement has been included, it is worth noting that in contrast DAPLink 0234 seems to cope with this specific example correctly and flashes all areas as expected without showing any errors.

flit, 2018-10-02:
@brianesquilona I'm surprised #503 didn't fix this problem. Any idea what the difference is?

brianesquilona, 2018-10-02:
This is definitely worth checking but I can see tat the the end address was detected by the decoder, so it did not continue.

brianesquilona, 2018-10-03:
@microbit-carlos I have added #510, but these changes will be reviewed for side effects. Can you help verify with your hex files. Thanks

microbit-carlos, 2018-10-05:
Hi @brianesquilona, yes I can test some of the hex files's we've got (the only odd ordering is this UICR area I've mentioned here). Is there any specific pattern/ordering in the hex file you want to test?

brianesquilona, 2018-10-09:
Thanks @microbit-carlos, you can verify your hex files that met your requirements. Just a note, non-ascending address support will trigger a sector setup, which may erase a page if supported, or writes 0xFF on addresses on sector boundary if the input address given is not on sector boundary.

microbit-carlos, 2018-10-11:
Apologies @brianesquilona, I haven't been able to test this yet and I will be away for a few days. I hope this can wait until Tuesday when I'm back.

brianesquilona, 2018-10-12:
@microbit-carlos it's fine, PR will always be there :)

PR #510 (2019-03-04): Added non-sequential non-increasing address support
Removed address > flash_end as end of flashing. Non-increasing address will cause a sector setup, page erase if supported.

closes #506
closed 2019-03-04 by brianesquilona