ARMmbed/DAPLink #512: NRF fails to program with "The interface firmware FAILED to reset/halt the target MCU"

studavekar, 2018-10-10:
steps to reproduce 

1. copy a test binary on to mount point `cp blinky.hex  /mnt/pci-0000_00_14_0-usb-0_5_2_1_1_0-scsi-0_0_0_0`

2. reset the deice during copy FAIL.TXT is created 
FAIL.TXT
```
(mbed-os-env) arm@austin-ci-linux-002:~/repo/nrf52_hwfw/mbed-os$ cat /mnt/pci-0000_00_14_0-usb-0_5_2_1_1_0-scsi-0_0_0_0/FAIL.TXT
error: The interface firmware FAILED to download the flash data contents to be programmed
type: target
```

3. try to copy the blinky.hex back to device `cp blinky.hex  /mnt/pci-0000_00_14_0-usb-0_5_2_1_1_0-scsi-0_0_0_0`

FAIL.TXT
```
(mbed-os-env) arm@austin-ci-linux-002:~/repo/nrf52_hwfw/mbed-os$ cat /mnt/pci-0000_00_14_0-usb-0_5_2_1_1_0-scsi-0_0_0_0/FAIL.TXT
error: The interface firmware FAILED to reset/halt the target MCU
type: target
```

brianesquilona, 2018-10-11:
@studavekar please try #513

studavekar, 2018-10-12:
Thanks, @brianesquilona this fixes the issue !

closed 2018-10-16 by studavekar