ARMmbed/DAPLink #516: RZ_A1H Flash algorithm write command FAILURE seen when automation turned on

studavekar, 2018-10-24:
DETAILS.TXT 
```
 DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 5500000003e0fb7200000000000000000000000097969902
HIC ID: 97969902
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0250
Git SHA: 6fece7592b9f0a57c032e48bfb70bd53d02dc33c
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Interface CRC: 0x74181e66
Remount count: 1
URL: https://mbed.org/device/?code=5500000003e0fb7200000000000000000000000097969902?version=0250?target_id=@T
```

FAIL.TXT
```
error: Flash algorithm write command FAILURE
type: target
```

Steps to reproduce 
1 ) Turn on the automation mode
2) copy a fimrware binary on device. It fails to program.

toyowata, 2018-10-29:
cc @ARMmbed/team-renesas @tomoyamanaka

TomoYamanaka, 2018-10-29:
We are under the investigation of the causes. It seems that the read error occurs at validation processing after writing.

TomoYamanaka, 2018-11-06:
We addressed this issue with PR #525, so could you check after it was merged?

studavekar, 2018-11-06:
Tested it locally fixes the issue.

closed 2018-12-28 by brianesquilona