ARMmbed/DAPLink #521: Flashing hanged when automation mode is on for testing (Toshiba TMPM366)

ganesh-ramachandran, 2018-11-02:
## Issue:

We are trying to test DAPLink by running below command:
`python test\run_test.py --force --firmware tmpm366_tmpm066_stk_if --firmwaredir projectfiles\uvision --testdl --loadbl --notestendpt`

If automation mode is enabled 
- Python test script not able to copy `tmpm366_bl_crc.hex` file (Hanged)

If disabled then
- test is failing, below is the log:

```
(venv) F:\DAPLink-0250>python test\run_test.py --force --loadbl --notestendpt --verbose All
[1541151968.159000] SubTest: DAPLink
[1541151968.322000] SubTest: Build test configuration
[1541151968.322000] No board to test firmware k20dx_frdmk64f_if
[1541151968.322000] Test configurations to be run:
[1541151968.322000]     0: APP=tmpm366_tmpm066_stk_if BL=tmpm366_bl Board=AdBun-M066 Target=<None>
[1541151968.322000]
[1541151968.338000]
[1541151968.338000] SubTest: tmpm366_tmpm066_stk_if AdBun-M066
[1541151968.338000] Board: Name=AdBun-M066 Unique ID=70110000412a381f00000001000000000000000097969909
[1541151968.338000] Application: Name=tmpm366_tmpm066_stk_if
[1541151968.338000] Bootloader: Name=tmpm366_bl
[1541151968.338000] Target: None
[1541151968.338000] SubTest: load_bootloader
[1541151968.338000] SubTest: set_mode
[1541151972.210000] programming took 3.66199994087 s
[1541151972.210000] SubTest: wait_for_remount
[1541151973.573000] unmount took 1.36299991608 s
[1541151975.648000] mount took 2.07500004768 s
[1541151975.648000] SubTest: set_mode
[1541151975.648000] Bootloader crc: 0x21b03983
[1541151975.648000] Bootloader CRC is wrong
[1541151975.648000] SubTest: load_interface
[1541151975.648000] SubTest: set_mode
[1541151975.648000] changing mode IF -> BL
[1541151975.663000] SubTest: wait_for_remount
[1541151976.504000] unmount took 0.84099984169 s
[1541151980.492000] mount took 3.95700001717 s
[1541151993.707000] programming took 12.4149999619 s
[1541151993.707000] SubTest: wait_for_remount
[1541151994.419000] unmount took 0.695999860764 s
[1541151997.137000] mount took 2.68000006676 s
[1541151997.137000] SubTest: set_mode
[1541151997.137000] Interface crc: 0x78b713d4

Test: tmpm366_tmpm066_stk_if AdBun-M066: Failure
  Info: Board: Name=AdBun-M066 Unique ID=70110000412a381f00000001000000000000000097969909
  Info: Application: Name=tmpm366_tmpm066_stk_if
  Info: Bootloader: Name=tmpm366_bl
  Info: Target: None
  Test: load_bootloader: Failure
    Test: set_mode: Pass
    Info: programming took 3.66199994087 s
    Test: wait_for_remount: Pass
      Info: unmount took 1.36299991608 s
      Info: mount took 2.07500004768 s
    Test: set_mode: Pass
    Info: Bootloader crc: 0x21b03983
    Failure: Bootloader CRC is wrong
  Test: load_interface: Pass
    Test: set_mode: Pass
      Info: changing mode IF -> BL
      Test: wait_for_remount: Pass
        Info: unmount took 0.84099984169 s
        Info: mount took 3.95700001717 s
    Info: programming took 12.4149999619 s
    Test: wait_for_remount: Pass
      Info: unmount took 0.695999860764 s
      Info: mount took 2.68000006676 s
    Test: set_mode: Pass
    Info: Interface crc: 0x78b713d4

Test Failed
```
**Steps to reproduce**

1 ) Download master from https://github.com/ganesh-ramachandran/DAPLink (since Toshiba TMPM366 is yet to be merged)
2 ) Generate the keil project and build
3 ) Turn on the automation mode
4)  Copy a firmware file on device. It fails to program and hanged.

Could you please suggest something we are missing?

brianesquilona, 2019-01-04:
@ganesh-ramachandran , the test is saying that the CRC in the bootloader image is not equal to the crc listed in the details.txt. The hanging of the hex file might be a different issue cause by absence of new line #510

ganesh-ramachandran, 2019-01-22:
Hi @brianesquilona, 
>the test is saying that the CRC in the bootloader image is not equal to the crc listed in the details.txt.

How can we manually verify that one? or Can you suggest where are we doing wrong, because as per porting guide we have added the board and the target. 

>The hanging of the hex file might be a different issue cause by absence of new line #510 

We tried with new line still hanging is happening.
The hanging is only happening if Automation mode is "ON", else without Automation mode test is failing.

brianesquilona, 2019-01-24:
@ganesh-ramachandran, you can check the crc in details.txt compared it to the crc when you compile the interface or bootloader bin, the post build script will add something like 

Start 0x0, Length 0xf000, CRC32 0x770a29e5

by tools/post_compute_crc.py

ganesh-ramachandran, 2019-03-27:
Hi @flit, @brianesquilona,

In bootloader case only, CRC is different in details.txt and post-build script CRC.

```
Post build script CRCs:
Start 0x0, Length 0xe000, CRC32 0x9ccec9bd : Bootloader
Start 0x10000, Length 0x17000, CRC32 0x2fbd0ace : DAPLink

Details.txt CRCs:
Bootloader CRC: 0xa751053b
Interface CRC: 0x2fbd0ace
```

Please suggest.

closed 2021-06-17 by mathias-arm