ARMmbed/DAPLink #536: Feature Request: Option to Disable the USB Disk

awneil, 2018-11-12:
Copied from https://github.com/mbedmicro/CMSIS-DAP/issues/98

The "USB disk" is great when you just have a .bin file you want to drop onto the target.

But, when doing "conventional" development - downloading & debugging via the debug interface - I find it annoying to have the "USB disk" keep popping up every time the board is restarted.

It would be useful to be able to disable it when not wanted.

therealprof, 2018-11-14:
I fully agree, especially when moving between boards having to eject the MSD (or ack the warning) can get a bit long in the tooth.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

awneil, 2018-11-14:
@theotherjimmy  - Thanks for your support!

Although I acknowledged in the OP that I do see the point of the MSD, it is a feature that I very rarely (if ever) use. I (almost) invariably do "conventional" development - so the MSD is  (almost) invariably a nuisance.

flit, 2018-11-14:
You aren't the only ones who are asking for this. We are taking this request seriously, and plan to enable a build without MSD or flash programming.

awneil, 2018-11-15:
@flit - thanks.

So the plan is for a build-time option - rather than a run-time configuration?

I guess the downside of that is ensuring that the board makers provide both builds to end users?

For completeness (unlikely to be an option that I would ever want), would there also be an option to disable the virtual COM port?

cederom, 2018-11-16:
Configuration like this could be done in runtime in bootloader and stored
for future boots.. then user could change the features in a safe place and
only one firmware would be supplied..?

--
CeDeROM, SQ7MHZ, http://www.tomek.cedro.info

PR #559 (2019-02-27): No-built-in-target implementation for Daplink
1. No flash algo builds, board IDs, family IDs can be inserted on post build.
2. MSD support can be turned off from the bootloader. MSD is automatically turned off on no flash algo builds.
3. All families built in builds for hics.

closes #536
closed 2019-02-27 by brianesquilona