ARMmbed/DAPLink #561: 251 version microbit firmware build issue

tanhailong, 2019-01-03:
Hi 
    Today I download the 0251 version of DAPLINK firmware,Then compile the microbit firmware,But it's compilation always reports the following error，How can i modify it ?Please help me,thank you.
SVC_Table.s: Error: C3472E: No source language specified (--cpp or --c90) and sourcefile extensions conflict
SVC_Table.s: error: A1905U: Pre-processor step failed for '..\..\..\source\daplink\interface\SVC_Table.s'
".\build\kl26z_microbit_if.axf" - 2 Error(s), 1 Warning(s).

brianesquilona, 2019-01-03:
@tanhailong please refer to [https://github.com/ARMmbed/DAPLink/blob/master/docs/DEVELOPERS-GUIDE.md](https://github.com/ARMmbed/DAPLink/blob/master/docs/DEVELOPERS-GUIDE.md), then please let me know where exactly the failure is.

tanhailong, 2019-01-05:
@brianesquilona  I use MDK version 5 it can be compiled and passed,But use MDK version 4 it all report 
this errors,I can assume that the latest version can only be compiled with a high version of MDK？

brianesquilona, 2019-01-07:
@tanhailong this maybe more connected to the arm compiler version, although MDK version 5 is now used more. You can try the mbedcli way of compiling binaries if you're not dependent on the IDE.

closed 2023-02-04 by mbrossard