ARMmbed/DAPLink #562: LPC546XX sector writes issues which causes bootloader update failure

brianesquilona, 2019-01-03:
A sequence of flash_erase_sector and flash_program_page in [critical_erase_and_program](https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/drag-n-drop/iap_flash_intf.c#L473) will produce a mismatch in the binary being written (can be checked by LPCScrypt_CLI.cmd ).

Another issue is [programming a memory from rom location](https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/drag-n-drop/iap_flash_intf.c#L460) will cause a fault, and will revert the interface chip to an LPC USB device, to be flashed by LPCScrypt.

brianesquilona, 2019-01-03:
@mmahadevan108 if you can check this issue for this HIC.

mmahadevan108, 2019-01-09:
Can you please provide steps to reproduce this failure.

brianesquilona, 2019-01-09:
@mmahadevan108 on interface mode, we should be able to update the bootloader, on a LPCXpresso546XX, you can try drag n drop a _bl (lpc4322_bl_crc.bin) file or bootloader binary for lpc4322.

mmahadevan108, 2019-01-09:
I am unable to drag-n-drop the bootloader binary. I see a FAIL.TXT file created on drag-n-drop with the below message.

The application file format is unknown and cannot be parsed and/or processed.

 I am using the Interface firmware from the 251 release.
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 105600001984a78e00000000000000000000000097969905
HIC ID: 97969905
Auto Reset: 0
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0251
Bootloader Version: 0244
Git SHA: 9295000cf241a1ac65236bb8cd3e30fa15310436
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x68b2eac9
Interface CRC: 0x076c30cb
Remount count: 1
URL: https://os.mbed.com/platforms/LPCXpresso54608/

mmahadevan108, 2019-01-11:
cc @flit  
After debugging I noticed a few issues in the implementation
1. The code fails when executing the below line
https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/drag-n-drop/iap_flash_intf.c#L460

It is passing in a Flash location as write buffer instead of RAM address, this causes the LPC program algorithm to fail.

2. I am not clear why there is an erase and program done on Line 460, my guess is you would like to intercept the erase so that is done during the final program on line 422 https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/drag-n-drop/iap_flash_intf.c#L422 

3. Line 422 is intending to program a sector however only 1 page is written, can you please clarify.

brianesquilona, 2019-01-11:
@mmahadevan108 
1. Yes, this is easy to fix.

2. L460 is a precaution, in case something bad happened during programming of bootloader, it will revert to your interface program, the intent is just for vectors, but flash has a minimum  write size.

3.  L422 will program a sector starting from the bootloader address, this was done at the end for the same purpose as 2. Although it will call flash_program_page, the size is for the whole sector.