ARMmbed/DAPLink #572: Turn page programming/sector erase to default

brianesquilona, 2019-01-28:
Current implementation defaults to erasing the whole flash, unless the page programming or sector erase is enabled by flash_manager_set_page_erase(true);

I think we should enable this one by default if the flash algo has support for erase_sector. If there are no objections :), I will create a PR for this.

therealprof, 2019-01-28:
I think this has both up- and downsides. On one hand using sector erase would not wipe out unprotected bootloaders and other data by accident. On the other hand partial erase has led to leakage of sensitive information more than just once...

flit, 2019-03-08:
@therealprof DAPLink is used in development environment, not for production firmware updates. So there should be no real secrets on the device. A device that has security requirements for managing sensitive information or keys must have debug access locked out in the production lifecycle state, so DAPLink wouldn't be able to program flash, anyway.

therealprof, 2019-03-08:
> DAPLink is used in development environment, not for production firmware updates. So there should be no real secrets on the device.

In past times we certainly have used debug probes for production purposes.

> A device that has security requirements for managing sensitive information or keys must have debug access locked out in the production lifecycle state, so DAPLink wouldn't be able to program flash, anyway.

Fair enough, though I wouldn't count on it that they always exist.

flit, 2019-03-09:
You're absolutely right, DAPLink can and will be used in production environments. However, there's a difference between factory production and end user firmware updates. The latter is what I was talking about.