ARMmbed/DAPLink #578: FRDM-KW24D512 un-responsive after Bricking in Windows10

sgrmtb, 2019-02-04:
Board:
      - FRDM-KW24D512
Problem:
      - Out of my four above types of boards, three became unresponsive due to bricking in Windows 10 or while trying to unbrick the bricked ones.or
Condition:
      - When Powering through the OpenLink USB No LED Glows or blinks
      - No drive appears in the Windows (while keeping the reset button pressed or otherwise)
      - When powering through the KW24 USB Two RED LED's Glow
What did I Try:
      - Followed the steps mentioned in DAPLink bootloader update | Mbed  regarding the problem and possible solutions using External Debug Probe (ULINK Pro).
      - Downloaded the DPLink Source from DAPLink/DEVELOPERS-GUIDE.md at master · ARMmbed/DAPLink · GitHub and compiled the solution as per the given instructions correctly for the k20dx_frdmkw24d_if as the board used OpenSDA MCY named K20DX128Vxx5. 
      - I used the following settings for the Options for the Target:
[image]
[image]
[image]

- The uVision solution compiled correctly without any error, see the Build Window below:

[image]

- I Flashed the Program, It again worked without any error. See the Build Window Below.

[image]

However, the Board did not come out of the Bricking State. THE SAME Problem Persists.

What NEXT?

brianesquilona, 2019-02-04:
@sgrmtb, you can also update the bootloader by compiling k20dx_bl, then flash it like you did from uvision, then try if you can get into the bootloader mode, then flash the interface software.

sgrmtb, 2019-02-04:
Hi, Brain,

Thanks for the concern, I will just try that. 

I have also tried another solution which also did not work.

What did I Try as the second method:
 
A. Copied the Bootloader from the Working Board using ULINK Pro into a Hex File.
   - Created a uVision Project with the K20DX128Vxx5.        
Used Save Command as follows to Read the Bootloader into a Hex FIle
SAVE MyFile.hex 0X0, 0X0001FFFF 
The hex file was created successfully in the project folder.
 
B. Flashed the Bricked boards with the copied Hex File.
   - Generate a uVision project
   - Select ULINKpro as debug interface
-   Entered the hex file name into 'Name of Executable' in the dialog 'Options for Target - Output'    
   - Performed a Flash download.
 
It showed that the programme has been flashed successfully, but
 
 
Same Problem Flashing Successful but the Boards not Working. AM I using the addresses correctly?

Meanwhile I will try with your solution.

Regards
Dr. M. Tariq Banday

e-mail:  sgrmtb@yahoo.com <mailto:sgrmtb@yahoo.com> (p), sgrmtb@gmail.com <mailto:sgrmtb@gmail.com> (a)
Url: http://banday.in <http://banday.in/>, Skype: tariq.banday

brianesquilona, 2019-02-04:
@sgrmtb the strange thing is that bootloader or MAINTENANCE mode/drive should be working (when you push the reset button and reconnect the usb). All the addresses you want to know is in daplink_addr.h.

You might also try Erase sector setting instead of erasing the whole chip.

sgrmtb, 2019-02-04:
Dear Brain,
	
	Thank You. It worked fine. I was able to get the board out of the bricking state. Thanks.

	I am still wondering why did the other solution not work.

	Further, I still have some problem with the debugging using ULINK Pro and flashing the program through ULINK Pro using uVision. 

	I have exported a program from the mBed and it compiles well in Keil. How should I flash into the main controller using ULINK Pro, so that I can use ULINK Pro as debugger.

Best Wishes

Dr. M. Tariq Banday
Head, Department of Electronics and Inst. Technology
University of Kashmir, Srinagar, 190006, J&K
Tel/Fax: +91 194 2415542 (direct) 
Tel: +91 194 2272381, Mob: +91 9419548922
e-mail:  sgrmtb@yahoo.com <mailto:sgrmtb@yahoo.com> (p), sgrmtb@gmail.com <mailto:sgrmtb@gmail.com> (a)
Url: http://banday.in <http://banday.in/>, Skype: tariq.banday

> On 05-Feb-2019, at 3:03 AM, Brian Esquilona <notifications@github.com> wrote:
> 
> daplink_addr.h

sgrmtb, 2019-02-04:
> @sgrmtb <https://github.com/sgrmtb> , programming the interface should not erase the bootloader, this might be the cause when you erased the flash on interface bin flashing.
>
> I'm not sure about the board but I think your ULINK is only connected to the interface chip, to connect to the main chip you can use the DAPLink via USB HID, uvision should be able to detect it.

Yes, it is right. I am able to debug using USB. I want to debug and program using ULINK Pro.

The board details are here:

https://www.nxp.com/docs/en/user-guide/FRDMKW24DUG.pdf <https://www.nxp.com/docs/en/user-guide/FRDMKW24DUG.pdf>

Do I have to connect my 10 pin cortex header to KW24SWD>

Best Wishes

Dr. M. Tariq Banday

e-mail:  sgrmtb@yahoo.com <mailto:sgrmtb@yahoo.com> (p), sgrmtb@gmail.com <mailto:sgrmtb@gmail.com> (a)
Url: http://banday.in <http://banday.in/>, Skype: tariq.banday

brianesquilona, 2019-02-04:
Yes, you already have the KW24 SWD pins exposed, just connect to this instead of the OpenLink Debug pins.

sgrmtb, 2019-02-04:
Yes, I did that, I am able to do it now.

I can also program it through this header?

Dr. M. Tariq Banday

e-mail:  sgrmtb@yahoo.com <mailto:sgrmtb@yahoo.com> (p), sgrmtb@gmail.com <mailto:sgrmtb@gmail.com> (a)
Url: http://banday.in <http://banday.in/>, Skype: tariq.banday

brianesquilona, 2019-02-04:
Yep, certainly. :)

sgrmtb, 2019-02-04:
Yes. Thanks.

With the same Board, Can I,

-  use ARM ULINK Plus for Energy Analysis in the similar manner or I have to cut some circuit?

-  use ARM ULINK Pro  and ULINK Plus in Kinetics Design Studio as well?

Dr. M. Tariq Banday

e-mail:  sgrmtb@yahoo.com <mailto:sgrmtb@yahoo.com> (p), sgrmtb@gmail.com <mailto:sgrmtb@gmail.com> (a)
Url: http://banday.in <http://banday.in/>, Skype: tariq.banday

closed 2021-06-17 by mathias-arm